analogdevicesinc/pyadi-iio #19: Accept floating point value or improve error message

eblot, 2019-09-16:
frequency should be specified as int, which prevents to specify it with exponentiation syntax, *i.e.*:

    $ PYTHONPATH=~/Sources/Git/github.com/pyadi-iio python3 library/py/cxi/pluto/plutotx2.py
    Traceback (most recent call last):
      File "library/py/cxi/pluto/plutotx2.py", line 14, in <module>
        sdr.rx_lo = 915e6
      File "/Users/eblot/Sources/Git/github.com/pyadi-iio/adi/ad9361.py", line 144, in rx_lo
        self._set_iio_attr("altvoltage0", "frequency", True, value)
      File "/Users/eblot/Sources/Git/github.com/pyadi-iio/adi/attribute.py", line 56, in _set_iio_attr
        raise ex
      File "/Users/eblot/Sources/Git/github.com/pyadi-iio/adi/attribute.py", line 54, in _set_iio_attr
        channel.attrs[attr_name].value = str(value)
      File "/usr/local/lib/python3.7/site-packages/iio.py", line 388, in <lambda>
        value = property(lambda self: self.__read(), lambda self, x: self.__write(x),
      File "/usr/local/lib/python3.7/site-packages/iio.py", line 404, in _Attr__write
        _c_write_attr(self._channel, self._name_ascii, value.encode('ascii'))
      File "/usr/local/lib/python3.7/site-packages/iio.py", line 44, in _checkNegative
        raise OSError(-result, _strerror(-result))
    OSError: [Errno 22] Invalid argument

I'm not sure if it would be legit to accept a float (and round it). However it would be nice to test the value to ensure it is a valid integer (`isinstance(value, int)`), as the traceback is not straightforward to understand.

tfcollins, 2019-09-16:
Casting would allow a user to provide incorrect inputs without the user knowing, which we want to avoid.

We could add some specific checking for integers to make the errors clearer

tfcollins, 2019-11-22:
Addressed here: https://github.com/analogdevicesinc/pyadi-iio/commit/604330f83eec248f97984771825a8da78e4584b5

closed 2019-11-22 by tfcollins