analogdevicesinc/pyadi-iio #39: Not all devices with "altvoltageX" channels are DDS devices

mihaiolteanu, 2019-12-06:
I have a `pulse-capture` device for Lidar, that looks like this,
```
iio:device5: 7c700000.axi-pulse-capture
		1 channels found:
			altvoltage0:  (output)
			3 channel-specific attributes found:
				attr  0: en value: 0
				attr  1: frequency value: 50000
				attr  2: pulse_width_ns value: 20
		6 device-specific attributes found:
				attr  0: sequencer_auto_cfg value: 0 1 2 3
				attr  1: sequencer_en value: 0
				attr  2: sequencer_manual_chsel value: 0 0 0 0
				attr  3: sequencer_mode value: auto
				attr  4: sequencer_mode_available value: auto manual
				attr  5: sequencer_pulse_delay_ns value: 248
		1 debug attributes found:
				debug attr  0: direct_reg_access value: 0x10061
```

In implementing a class for this device, I want to inherit from the `tx` class. The constructor for the `tx` class also does a `dds.__init__(self)` since it inherits from `dds`. So far so good. The `dds` constructor then searches for all `altvoltageX` channels and sets the `frequency`, `scale`, `phase` and `raw` attributes for each of them. I'm assuming that this naming convention is seen on DDS devices (whatever those are) so you are assuming what type of device you have based on the channel names.

In the case of the `pulse-capture` device from above, on the other hand, there is indeed an `altvoltage0` channel, but only the `frequency` attribute is present and the rest of the attributes, `scale`, `phase` and `raw` are missing so I get a `KeyError` for all of these attributes.  The setting of `frequency` actually fails, too, with an `invalid argument` error string, since `0` is not a valid value to set the `frequency` to.

tfcollins, 2019-12-06:
All our HDL designs, which hook up to ADI devices, with transmitters have DDS cores inside them. This is what the dds.py class controls.

LIDARs do not typically have buffers so the tx.py class might not be that useful. What driver and device are you using? We might be able to add some enhancements to the core class to make this easier for you.

mihaiolteanu, 2019-12-06:
> LIDARs do not typically have buffers

What do you mean by this? You are right regarding this particular device, in that I don't have a scenario in which I need to send something to the `axi-pulse-capture`, so there is no need for a buffer. The only scenario for this `axi-pulse-capture` device is the one in which I need to configure it (i.e. write some attributes). The approach, at the moment, and not necessarily the correct one,  is to define a `pulse-capture` class that inherits from `tx`, add it to the list of supported devices for this library, and use it by the Lidar class. 

> What driver and device are you using?

I'm not really sure to what device this `axi-pulse-capture` belongs. From what I understand this device is part of the FPGA board itself?! The list of devices as returned by iiolib for the Lidar board is `ad7091, ltc2471, xadc, ad5627, ad9528, 7c700000.axi-pulse-capture, axi-ad9094-hpc`

I'm off work now, but I'll try any suggestions you might have once I get my hands on a Lidar board.

Thanks for your support,
Mihai

tfcollins, 2019-12-06:
> What do you mean by this? You are right regarding this particular device, in that I don't have a scenario in which I need to send something to the axi-pulse-capture, so there is no need for a [...]

Yes this is basically what I mean. However, the current tx really doesn't apply to LIDAR since the purpose of the class is to handle IIO devices with scan elements (data buffers).  `axi-pulse-capture` is pure control logic, so you could use it similar to how to have _ctrl objects for the other classes.

>I'm not really sure to what device this axi-pulse-capture belongs. From what I understand this device is part of the FPGA board itself?! The list of devices as returned by iiolib for the Lidar board is ad7091, ltc2471, xadc, ad5627, ad9528, 7c700000.axi-pulse-capture, axi-ad9094-hpc

I guess this is the AD-FMCLIDAR1-EBZ board. Based on how that board works you should just have properties to set the laser pulse generator settings and not require a specific class. The only necessary pieces where you would have a specific generic say lidar class would be be triggered captures. I would just inherit from rx and override the rx method to call a trigger before calling the superclass rx method.

closed 2019-12-09 by mihaiolteanu