analogdevicesinc/pyadi-iio #42: Example code not working

sunipkmukherjee, 2019-12-20:
I have installed adi through pip on Python 3.6.9 and I have libiio 0.18 (git tag f69d201).
There are two specific issues when I try to execute the following code:

> import adi (success)
> sdr = adi.ad9361() 
Fails, with:
ERROR: Unable to find host: Name or service not known
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/adi/context_manager.py", line 56, in __init__
    self._ctx = iio.Context(self._uri_auto)
  File "/usr/local/lib/python3.6/dist-packages/iio.py", line 736, in __init__
    self._context = _new_uri(_context.encode('ascii'))
  File "/usr/local/lib/python3.6/dist-packages/iio.py", line 33, in _checkNull
    raise OSError(err, _strerror(err))
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/adi/ad9361.py", line 48, in __init__
    context_manager.__init__(self, uri, self._device_name)
  File "/usr/local/lib/python3.6/dist-packages/adi/context_manager.py", line 62, in __init__
    raise Exception("No device found")
Exception: No device found

Circumvented using:
> sdr = adi.ad9361('local:')

Then when I try to change an attribute, say
> sdr.rx_rf_bandwidth = 4000000, it fails with:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/adi/ad9361.py", line 117, in rx_rf_bandwidth
    self._set_iio_attr_int("voltage0", "rf_bandwidth", False, value)
  File "/usr/local/lib/python3.6/dist-packages/adi/attribute.py", line 71, in _set_iio_attr_int
    self._set_iio_attr(channel_name, attr_name, output, value, _ctrl)
  File "/usr/local/lib/python3.6/dist-packages/adi/attribute.py", line 57, in _set_iio_attr
    raise ex
  File "/usr/local/lib/python3.6/dist-packages/adi/attribute.py", line 55, in _set_iio_attr
    channel.attrs[attr_name].value = str(value)
  File "/usr/local/lib/python3.6/dist-packages/iio.py", line 376, in <lambda>
    value = property(lambda self: self.__read(), lambda self, x: self.__write(x),
  File "/usr/local/lib/python3.6/dist-packages/iio.py", line 392, in _Attr__write
    _c_write_attr(self._channel, self._name_ascii, value.encode('ascii'))
  File "/usr/local/lib/python3.6/dist-packages/iio.py", line 39, in _checkNegative
    raise OSError(-result, _strerror(-result))
OSError: [Errno 22] Invalid argument

Please advise if you can.

tfcollins, 2019-12-20:
What device do you have?

sunipkmukherjee, 2019-12-23:
It's an ADRV9364-Z7020 eval board.

sunipkmukherjee, 2019-12-23:
It definitely contains an `ad9361` device, which shows up on `iio_info` and works as intended using the `ad9361_iiostream.c` example code provided with `libiio`.

tfcollins, 2019-12-23:
`ADRV9364-Z7020` contains an AD9364 not AD9361. The channel count is different. Please try adi.ad9364.

sunipkmukherjee, 2019-12-23:
Thank you, `adi.ad9361('local:')` works in an AD9361 board. I wonder why the generic instantiation is not working though.

tfcollins, 2019-12-24:
The AD9361 class tries to initialize 2 RX and 2 TX channels, which won't exist on AD9364 based devices.

tfcollins, 2020-01-08:
Can I close this?

closed 2020-01-08 by sunipkmukherjee