analogdevicesinc/pyadi-iio #55: Issues with importing adi in Python after install

The-inside-man, 2020-01-29:
Sorry to bring this up again.. 

I worked with the folks in the Gnuradio slack chat and got everything installed properly finally with all necessary packages. Everything seems to be good to go now, however when I try to import adi in python I get the following error:

jake@jake-Blade:~/pyadi-iio$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import iio
>>> import adi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jake/pyadi-iio/adi/__init__.py", line 34, in <module>
    from adi.ad936x import *
  File "/home/jake/pyadi-iio/adi/ad936x.py", line 35, in <module>
    from adi.rx_tx import rx_tx
  File "/home/jake/pyadi-iio/adi/rx_tx.py", line 43, in <module>
    class phy(attribute):
  File "/home/jake/pyadi-iio/adi/rx_tx.py", line 44, in phy
    _ctrl: iio.Device = []
AttributeError: module 'iio' has no attribute 'Device'
>>> exit()
[1]+  Killed                  import adi

Seems that iio is missing some attribute device. Is it possible there is a version issue with libraries where iio did not support "device" or perhaps something along that sort? You can see I successfully imported iio but seems to still be missing stuff...

Thank you as always.

Jake

tfcollins, 2020-01-29:
This is an issue with gr-iio and libiio's bindings having the same module name. With your setup `import iio` actually imports the swig wrapper for the gr-iio blocks. One workaround is to put the libiio binding in a folder that is searched before the gr-iio wrapper. Python typically goes in linear order through the list in PYTHONPATH. So the first one it finds it uses.

The-inside-man, 2020-01-29:
When adding libiio to my PYTHONPATH so it searches first it appears to find it, but finds errors within iio.py:

jake@jake-Blade:/usr/lib/python3.6/site-packages$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import iio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/iio.py", line 71, in <module>
    _get_backends_count = _lib.iio_get_backends_count
  File "/usr/lib/python3.6/ctypes/__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: python3: undefined symbol: iio_get_backends_count

tfcollins, 2020-01-29:
It's erroring on the first call that actually uses the .so. So the libiio .so is not installed or not on path.

tfcollins, 2020-02-07:
Has this been resolved?

The-inside-man, 2020-02-08:
I can import iio now but I do not see the FMComm blocks in GNU radio

tfcollins, 2020-02-08:
What is the output of `python3 -c "import iio; print(dir(iio))"`

The-inside-man, 2020-02-09:
jake@jake-Blade:~$ python3 -c "import iio; print(dir(iio))"
['DEFAULT_BUFFER_SIZE', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '_iio_pluto_sink_swig', '_iio_pluto_source_swig', '_iio_swig', 'attr_sink', 'attr_sink_make', 'attr_sink_sptr', 'attr_sink_sptr_swigregister', 'attr_sink_swigregister', 'attr_source', 'attr_source_make', 'attr_source_sptr', 'attr_source_sptr_swigregister', 'attr_source_swigregister', 'attr_updater', 'device_sink', 'device_sink_make', 'device_sink_make_from', 'device_sink_sptr', 'device_sink_sptr_swigregister', 'device_sink_swigregister', 'device_source', 'device_source_make', 'device_source_make_from', 'device_source_sptr', 'device_source_sptr_swigregister', 'device_source_swigregister', 'fmcomms2_sink', 'fmcomms2_sink_f32c', 'fmcomms2_sink_f32c_make', 'fmcomms2_sink_f32c_sptr', 'fmcomms2_sink_f32c_sptr_swigregister', 'fmcomms2_sink_f32c_swigregister', 'fmcomms2_sink_make', 'fmcomms2_sink_make_from', 'fmcomms2_sink_sptr', 'fmcomms2_sink_sptr_swigregister', 'fmcomms2_sink_swigregister', 'fmcomms2_source', 'fmcomms2_source_f32c', 'fmcomms2_source_f32c_make', 'fmcomms2_source_f32c_sptr', 'fmcomms2_source_f32c_sptr_swigregister', 'fmcomms2_source_f32c_swigregister', 'fmcomms2_source_make', 'fmcomms2_source_make_from', 'fmcomms2_source_sptr', 'fmcomms2_source_sptr_swigregister', 'fmcomms2_source_swigregister', 'fmcomms5_sink', 'fmcomms5_sink_f32c', 'fmcomms5_sink_f32c_make', 'fmcomms5_sink_f32c_sptr', 'fmcomms5_sink_f32c_sptr_swigregister', 'fmcomms5_sink_f32c_swigregister', 'fmcomms5_sink_make', 'fmcomms5_sink_make_from', 'fmcomms5_sink_sptr', 'fmcomms5_sink_sptr_swigregister', 'fmcomms5_sink_swigregister', 'fmcomms5_source', 'fmcomms5_source_f32c', 'fmcomms5_source_f32c_make', 'fmcomms5_source_f32c_sptr', 'fmcomms5_source_f32c_sptr_swigregister', 'fmcomms5_source_f32c_swigregister', 'fmcomms5_source_make', 'fmcomms5_source_make_from', 'fmcomms5_source_sptr', 'fmcomms5_source_sptr_swigregister', 'fmcomms5_source_swigregister', 'high_res_timer_epoch', 'high_res_timer_now', 'high_res_timer_now_perfmon', 'high_res_timer_tps', 'iio_math', 'iio_math_gen', 'iio_math_gen_make', 'iio_math_gen_sptr', 'iio_math_gen_sptr_swigregister', 'iio_math_gen_swigregister', 'iio_math_make', 'iio_math_sptr', 'iio_math_sptr_swigregister', 'iio_math_swigregister', 'iio_pluto_sink_swig', 'iio_pluto_source_swig', 'iio_swig', 'modulo_const_ff', 'modulo_const_ff_make', 'modulo_const_ff_sptr', 'modulo_const_ff_sptr_swigregister', 'modulo_const_ff_swigregister', 'modulo_ff', 'modulo_ff_make', 'modulo_ff_sptr', 'modulo_ff_sptr_swigregister', 'modulo_ff_swigregister', 'os', 'pluto_sink', 'pluto_sink_make', 'pluto_sink_sptr', 'pluto_sink_sptr_swigregister', 'pluto_sink_swigregister', 'pluto_source', 'pluto_source_make', 'pluto_source_sptr', 'pluto_source_sptr_swigregister', 'pluto_source_swigregister', 'power_ff', 'power_ff_make', 'power_ff_sptr', 'power_ff_sptr_swigregister', 'power_ff_swigregister', 'unicode_literals']

The-inside-man, 2020-02-09:
so clearly it is there... how do I get my gnuradio-companion to see all of it. How do I set its path?

tfcollins, 2020-02-09:
CMake should put the blocks here: https://github.com/analogdevicesinc/gr-iio/blob/upgrade-3.8/CMakeLists.txt#L103 

If they are not there, try reinstalling.

The-inside-man, 2020-02-09:
When I look in the location, there are two installations for grc... under /usr/local/share/gnuradio and again under /usr/share/gnuradio... both seem to have the iio library showing the iio blocks I need but in the gui itself, they do not appear.

tfcollins, 2020-02-10:
If blocks do not appear you can manually add them to the import path: https://wiki.gnuradio.org/index.php/GNURadioCompanion#Installing_the_XML_Block_Definition . This should apply to yml blocks as well.

The-inside-man, 2020-02-14:
I have updated my config.conf under ~/.gnuradio with the following:

jake@jake-Blade:~/.gnuradio$ grep "" config.conf
[grc]
local_blocks_path=/usr/local/share/gnuradio/grc/blocks:/usr/share/gnuradio/grc/blocks

I still do not see any of the blocks for iio in GRC...

if I run a locate:

/usr/share/gnuradio/grc/blocks/iio.tree.yml
/usr/share/gnuradio/grc/blocks/iio_attr_sink.block.yml
/usr/share/gnuradio/grc/blocks/iio_attr_source.block.yml
/usr/share/gnuradio/grc/blocks/iio_attr_updater.block.yml
/usr/share/gnuradio/grc/blocks/iio_device_sink.block.yml
/usr/share/gnuradio/grc/blocks/iio_device_source.block.yml
/usr/share/gnuradio/grc/blocks/iio_fmcomms2_sink.block.yml
/usr/share/gnuradio/grc/blocks/iio_fmcomms2_source.block.yml
/usr/share/gnuradio/grc/blocks/iio_fmcomms5_sink.block.yml
/usr/share/gnuradio/grc/blocks/iio_fmcomms5_source.block.yml
/usr/share/gnuradio/grc/blocks/iio_pluto_sink.block.yml

The-inside-man, 2020-02-14:
running the companion it has the correct paths:

Block paths:
	/home/jake/gr-iio/grc
	/usr/local/share/gnuradio/grc/blocks
	/usr/share/gnuradio/grc/blocks
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

tfcollins, 2020-02-14:
Does gnuradio companion produce any errors on startup?

The-inside-man, 2020-02-14:
no, only warnings for importing duplicate blocks. but I do not see the iio blocks part of these warnings, which makes me think they still are not being imported. Since I have them in two of the import locations.

The-inside-man, 2020-02-28:
Turns out there was two versions of Gnuradio installed in the end. After completely uninstalling both versions and then making from source for gnuradio 3.8 of the 3.8 branch, the issue went away.

closed 2020-02-28 by The-inside-man

FRiZoK, 2024-12-11:
Hello, I have a similar problem on windows.
I don't understand what my problem is, because I installed everything that was needed, but python doesn't find the file.
Can you help me?

  File "g:\Ycheba\Aleksey\4kyrs\Diplom\Pluto\Code\Pluto_revC_rev2.py", line 57, in <module>
    import adi
  File "c:\Users\leven\AppData\Local\Programs\Python\Python310\lib\site-packages\adi\__init__.py", 
line 5, in <module>
    from adi.ad2s1210 import ad2s1210
  File "c:\Users\leven\AppData\Local\Programs\Python\Python310\lib\site-packages\adi\ad2s1210.py", 
line 6, in <module>
    from adi.context_manager import context_manager
  File "c:\Users\leven\AppData\Local\Programs\Python\Python310\lib\site-packages\adi\context_manager.py", line 5, in <module>
    import iio
  File "g:\Ycheba\Aleksey\4kyrs\Diplom\Pluto\Code\iio.py", line 303, in <module>
    _lib = _cdll("libiio.so.1", use_errno=True, use_last_error=True)
  File "c:\Users\leven\AppData\Local\Programs\Python\Python310\lib\ctypes\__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'libiio.so.1' (or one of its dependencies). Try using the 
full path with constructor syntax.