analogdevicesinc/pyadi-iio #78: problems when I try to install pyadi-iio with python 3.5.2

JanVerlinden, 2020-04-01:
Is python 3.5.2. not supported? I have the impression that I need python 3.6? Or is there a simple work around to use pyadi-iio in python 3.5.2?

tfcollins, 2020-04-01:
pyadi-iio requires 3.6+ https://github.com/analogdevicesinc/pyadi-iio/blob/master/setup.py#L32

The main requirement is that 3.6 introduced annotations which we use heavily.

Can I ask why 3.5.2?

JanVerlinden, 2020-04-01:
Ubuntu 16.04.5 LTS (GNU/Linux 4.14.0+ aarch64)
The software modules that we have build are not compatible with higher versions.

tfcollins, 2020-04-01:
If you remove the annotations the module should function in 3.5. We use them for static analysis and self-documentation, so they do not change the functionality. We don't plan on supporting anything below 3.6 in the future since 3.5.9 is EOL in September.

tfcollins, 2020-04-30:
Closing for now

closed 2020-04-30 by tfcollins