analogdevicesinc/pyadi-iio #91: libiio: ProcessLookupError: [Errno 3] No such process

Deville-ee, 2020-05-26:
I'm having this error when I try to initialize an adrv9009 device with pyadi-iio 

> 

>$ sudo python3.7                     
Python 3.7.6 (default, May 25 2020, 13:57:06) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adi
 >>> import iio
>>> iio.version
(0, 19, 'v0.19')
>>> sdr=adi.adrv9009()
ERROR: Unable to find host: Temporary failure in name resolution
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/adi/context_manager.py", line 56, in __init__
    self._ctx = iio.Context(self._uri_auto)
  File "/home/zynqmp/libiio/bindings/python/iio.py", line 736, in __init__
    self._context = _new_uri(_context.encode('ascii'))
  File "/home/zynqmp/libiio/bindings/python/iio.py", line 33, in _checkNull
    raise OSError(err, _strerror(err))
ProcessLookupError: [Errno 3] No such process

>During handling of the above exception, another exception occurred:

>Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.7/site-packages/adi/adrv9009.py", line 48, in __init__
    context_manager.__init__(self, uri, self._device_name)
  File "/usr/lib/python3.7/site-packages/adi/context_manager.py", line 62, in __init__
    raise Exception("No device found")
Exception: No device found
>>> 
``

What steps should I take to debug this?  How is the device initialized/found?

Deville-ee, 2020-05-26:
Found it at: https://github.com/analogdevicesinc/libiio/blob/master/context.c#L299

So, we have a local adrv9009 device plugged into our board and in fact pyadi initializes the device with the default value ip:analog which throws an NULL error in the iio library. 

sdr=adi.adrv9009("local:")  solves this issue. 

I could not find , however, any info about this type of initialization in the documentation.

tfcollins, 2020-05-27:
This is documented here: https://analogdevicesinc.github.io/pyadi-iio/guides/connectivity.html

Deville-ee, 2020-05-28:
I don't see this on you linked page. 

 We used to iniitialize with: sdr=adi.adrv9009()   just like the example sdr = adi.Pluto()
but this does not seem to work, now we need to initialize with:

sdr=adi.adrv9009("local:")

We never used a parameter to find out target board. Now we have to.

tfcollins, 2020-05-28:
> sdr=adi.adrv9009("local:")

This is setting the URI to local, which is the same as `sdr=adi.adrv9009(uri='local:')`. See the constructor call https://analogdevicesinc.github.io/pyadi-iio/devices/adi.adrv9009.html

If you leave it blank it will try to automatically connect to one and will connect to the first one it finds. If you have multiple and it can connect to the wrong one, which is likely what's going on.

tfcollins, 2020-09-25:
I think this is resolved. Closing

closed 2020-09-25 by tfcollins