analogdevicesinc/pyadi-iio #92: Wish: conda package for pyadi-iio (windows)

petrushy, 2020-05-26:
Hi

It would be very helpful with a conda package for pyadi-iio including the dependency libiio as it seems quite difficult to get this integrated in python as a user on windows.

see conda-forge for more info on conda packaging

Thanks!

tfcollins, 2020-05-26:
Is this more related to libiio or pyadi-iio? Pip should be in every windows install of python.

Is Pip not part of your normal flow?

-Travis

petrushy, 2020-05-26:
Hi,

I might have got things wrong, but installing via pip seems only to install the pyadi-ioo library, not the needed dependencies.

Conda is a package manager that can distribute binary python (and other code) packages and libraries in contained environments, typically used in the anaconda python distribution for scientific usage. With this I would imagine it is possible to distribute it with the needed libiio. The windows installer will place the files on the system itself, not in a contained environment. And needs admin privilegies.

tfcollins, 2020-05-26:
Yes I'm familiar with conda. This issue is more related to libiio though since it doesn't have a pypi or other related installer (yet). This will happen in v0.20 which is slated for June. Once it's there we can look into more packaging options.

petrushy, 2020-05-27:
Great, yes it is related to libiio and that being separate as a dependency.

Good to hear it is planned! Many thanks!

closed 2020-05-27 by petrushy