analogdevicesinc/pyadi-iio #96: Change device finding to error when null returned

tfcollins, 2020-06-04:


engkan2kit, 2020-08-03:
Can you assign this to me? I'm going to add more tests on the other boards that we have so I'll probably have to add this to make use of this feature.

kister-jimenez, 2020-08-03:
I'm assigning this to my self. I'll be adding more boards to the tests so I'll be using this feature too.

vaidehisingh12, 2026-10-31:
Is this still wanted? I had a look and the classes built on `rx_def`/`tx_def` already raise "No device found with name ..." when `find_device()` returns `None`. It's the older classes that still don't check. For example, with an empty context `adi.ad9094()` and `adi.ad9136()` construct without any error, and `adi.ad9625()` / `adi.ad9250()` fail with `TypeError: 'NoneType' object is not callable`.

I'd be happy to pick it up, but I wanted to check how you'd want it done first, since there are a lot of call sites and I know those classes are being moved onto `device_base`:

1. A small helper on `context_manager` that raises the same message as `rx_tx.py`, used in the legacy classes a few at a time, or
2. Leave it to the `device_base` migration and close this.

(Also, it's still assigned from 2020. Happy to leave it if someone's on it.)