analogdevicesinc/pyadi-iio #97: Get/set kernel buffer count of ADALM-Pluto via Pyadi

xwkang2019, 2020-06-08:
Dear Developer,

Is there any way we could get/set the number of kernel buffers for ADALM-Pluto using Pyadi? I know such functions exist in libiio but I am not sure how I could do that via Pyadi after getting my context from "sdr = adi.Pluto()". Is there any command like "sdr.kernel_buffers_count = 32"? 

In addition, previously when I was working with ADALM-2000, there was a very nice documentation page that included all available APIs: https://analogdevicesinc.github.io/libm2k/annotated.html
Is there a similar page for PyADI? If not, would it be better for us to program directly using libiio if we would like to have access to more APIs in order to configure ADALM-Pluto as needed?

Thank you in advance for your help!
Steven

tfcollins, 2020-06-08:
Doc for pyadi: https://analogdevicesinc.github.io/pyadi-iio/
libiio python bindings: https://analogdevicesinc.github.io/libiio/master/python/index.html

To set the kernel buffers you have to access some of the core classes. For most device specific classes (and Pluto) this is:
```
sdr = adi.Pluto()
sdr._rxadc.set_kernel_buffers_count(<count>)
```
-Travis

tfcollins, 2020-06-16:
I think this is answered. Closing

closed 2020-06-16 by tfcollins