analogdevicesinc/pyadi-iio #128: Namespace Collision with gr-iio. Cannot import adi in python3.

Teque5, 2020-10-21:
In https://github.com/analogdevicesinc/pyadi-iio/issues/55 there is some discussion that having `gr-iio` installed (for GNU Radio) which normally imports as `iio` prevents `pyadi-iio` from working correctly. This issue still exists:

```python
In [1]: import adi
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-1bef210865df> in <module>
----> 1 import adi

~/.local/lib/python3.6/site-packages/adi/__init__.py in <module>
     32 # THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
     33
---> 34 from adi.ad936x import *
     35
     36 from adi.fmcomms5 import *

~/.local/lib/python3.6/site-packages/adi/ad936x.py in <module>
     33
     34 from adi.context_manager import context_manager
---> 35 from adi.rx_tx import rx_tx
     36
     37

~/.local/lib/python3.6/site-packages/adi/rx_tx.py in <module>
     41
     42
---> 43 class phy(attribute):
     44     _ctrl: iio.Device = []
     45

~/.local/lib/python3.6/site-packages/adi/rx_tx.py in phy()
     42
     43 class phy(attribute):
---> 44     _ctrl: iio.Device = []
     45
     46     def __del__(self):

AttributeError: module 'iio' has no attribute 'Device'
```

I believe this occurs since the `libiio` python bindings are installed to the same namespace as `gr-iio`. This namespace issue has occured in many other other projects (gr-sigmf & sigmf) and should be resolved by renaming one of the offending packages.

It makes sense to me that `libiio` should be used in python like `import libiio`.

If a proposed solution includes modifying `$PATH` or `$PYTHONPATH`, it is a __bad__ solution.

Let me know if I overlooked something. Using `pybombs` installed GNU Radio 3.8 & Python 3.6.9.

tfcollins, 2020-10-21:
We are in the process of merging gr-iio upstream into gnuradio itself, which will change the namespace to be gnuradio.iio. That is the long term plan in regards to this issue: https://github.com/gnuradio/greps/blob/master/grep-0017-iio.md

gr-iio in 3.7 actually put things in the gnuradio namespace which actually caused invalid installs a lot. That is why 3.8 moved things around.

You are free to change the bindings naming locally if you want but the iio python bindings can't really be changed at this point. They have been around before gr-iio existed, are already in major packages, and are used outside of the GR community.

-Travis

Teque5, 2020-10-21:
Ah yes I should have remembered the merge from the discussion at grcon. I suppose my solution to this will be to upgrade to GNU Radio 3.9 once you guys have finished.

Thanks for taking a look.

closed 2020-10-21 by tfcollins