analogdevicesinc/pyadi-iio #151: transmission not stopped, after python application closed

archy-embedinn, 2020-12-11:
i am using raw python API, using buffers in cyclic mode
       
        self.txbuf = iio.Buffer(self.txdac, self.samples_per_channel, False)
         ...
        self.txbuf.write(bytearray(iq_table))
        self.txbuf.push()

after the application closed a tone transmission still continue, how can i stop the transmission before the application exit

tfcollins, 2020-12-11:
What device are you using? and why are you using buffers directly?

-Travis

archy-embedinn, 2020-12-11:
I am using adrv9364,
I don't recall the issue I was facing and why I somehow switch to the low-level APIs, but for now, all my application is based on it.
can you please recommend anything to switch down the tone, I will add this in my application exit function. Thanks

tfcollins, 2020-12-11:
When you clear the context (close your app) the TX driver will switch to its default mode (DDS mode). A simple solution is to set the DDS scales to zero so you are passing zeros to the DAC:
```
sdr = adi.AD9364(<uri>)
...
sdr.dds_scales = [0,0,0,0]
del sdr
```
You can power down the TX side but if you want to use it again you will have to explicitly power it up.

-Travis

tfcollins, 2020-12-11:
One other thing:
```
self.txbuf = iio.Buffer(self.txdac, self.samples_per_channel, False)
```
This is not a cyclic buffer, False -> True to get cyclic.

-Travis

archy-embedinn, 2020-12-12:
Travis thanks very much, this trick works .....

        self.txdac.find_channel('altvoltage0',True).attrs['scale'].value = str(0)
        self.txdac.find_channel('altvoltage1',True).attrs['scale'].value = str(0)
        self.txdac.find_channel('altvoltage2',True).attrs['scale'].value = str(0)
        self.txdac.find_channel('altvoltage3',True).attrs['scale'].value = str(0)

 I also observe that the tone remains down even next time the app runs without the above code, I think it remains low as it zeroes the regs in the FPGA. Should not it be part of the driver's init or exit to damp the tone down, when the driver loads?

tfcollins, 2020-12-12:
The driver will enumerate in a default state a boot. If you want a certain behavior, like zeroing out the DDSs, I would recommend just adding it to your application's init.

-Travis

archy-embedinn, 2020-12-13:
OK, Thanks Travis
as a side note, I am amazed how fast you do the development gr-cons/fpga/python/linux etc, and along giving this quick support Thumps up

closed 2020-12-13 by archy-embedinn