analogdevicesinc/pyadi-iio #154: ZCU102-ADRV9009 tests fail on 4.19 with JESD FSM

flpm2, 2020-12-16:
I planned to reuse the zcu102 - adrv9009 unit tests in the current package as a sanity check during custom board development. 

On the latest master of pyadi-iio I see the following output:

```
-- Docs: https://docs.pytest.org/en/stable/warnings.html
========================================================================================================== short test summary info ==========================================================================================================
FAILED test_adrv9009_p.py::test_adrv9009_sfdr[45-param_set0-0-adi.adrv9009] - AssertionError
FAILED test_adrv9009_p.py::test_adrv9009_sfdr[45-param_set1-0-adi.adrv9009] - AssertionError
FAILED test_adrv9009_p.py::test_adrv9009_sfdr[45-param_set2-0-adi.adrv9009] - AssertionError
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_agc[0.9-0-22-param_set0-0-adi.adrv9009] - AssertionError
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_agc[0.9-0-22-param_set1-0-adi.adrv9009] - AssertionError
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_agc[0.9-0-22-param_set2-0-adi.adrv9009] - AssertionError
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-0-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-1-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-2-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-3-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-0-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-1-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-2-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-3-adi.adrv9009] - TypeError: gain_check() missing 1 required positional argument: 'max_rssi'
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-0-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-1-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-2-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-3-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-0-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-1-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-2-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-3-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-0-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-1-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-2-adi.adrv9009] - OSError: [Errno 5] Input/output error
FAILED test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-3-adi.adrv9009] - OSError: [Errno 5] Input/output error
```
Part of it is due to me having loopback only on Tx2-Rx2, but I'm wonderig about the OSError. Do you have any tips?

Envvironments:
On target:
 *  I'm using the latest and greatest ad-linux 4.19 (commit 61293bbe4a5b912d5f5913c3209bf2fb19a9ee63 (origin/adi-4.19.0) with the new JESD204 FSM device tree.
 

On host and target
 *  libiio master
 *  pyadi-iio master
 *  pytest-libiio master

Currently I execute commands on host ubuntu20.10 with 5.8.0-34-generic #37-Ubuntu SMP Thu Dec 10 18:01:14 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux.

tfcollins, 2020-12-16:
Can you give the full traceback from one of the OSError resulting tests?

-Travis

flpm2, 2020-12-17:
Hi Travis, 

Thank you for responding in such a short notice!

Here is a trace
```
___________________________________________________________________________________________ test_adrv9009_iq_loopback[param_set2-3-adi.adrv9009] ____________________________________________________________________________________________

test_iq_loopback = <function cw_loopback at 0x7f7285db6940>, iio_uri = 'ip:analog', classname = 'adi.adrv9009', channel = 3
param_set = {'calibrate': 1, 'calibrate_tx_qec_en': 1, 'gain_control_mode_chan0': 'slow_attack', 'gain_control_mode_chan1': 'slow_attack', ...}

    @pytest.mark.iio_hardware(hardware)
    @pytest.mark.parametrize("classname", [(classname)])
    @pytest.mark.parametrize("channel", [0, 1, 2, 3])
    @pytest.mark.parametrize(
        "param_set",
        [
            dict(
                trx_lo=1000000000,
                gain_control_mode_chan0="slow_attack",
                gain_control_mode_chan1="slow_attack",
                tx_hardwaregain_chan0=-20,
                tx_hardwaregain_chan1=-20,
                calibrate_tx_qec_en=1,
                calibrate=1,
            ),
            dict(
                trx_lo=3000000000,
                gain_control_mode_chan0="slow_attack",
                gain_control_mode_chan1="slow_attack",
                tx_hardwaregain_chan0=-20,
                tx_hardwaregain_chan1=-20,
                calibrate_tx_qec_en=1,
                calibrate=1,
            ),
            dict(
                trx_lo=5000000000,
                trx_lo_chip_b=1000000000,
                gain_control_mode_chan0="slow_attack",
                gain_control_mode_chan1="slow_attack",
                tx_hardwaregain_chan0=-20,
                tx_hardwaregain_chan1=-20,
                calibrate_tx_qec_en=1,
                calibrate=1,
            ),
        ],
    )
    def test_adrv9009_iq_loopback(test_iq_loopback, iio_uri, classname, channel, param_set):
>       test_iq_loopback(iio_uri, classname, channel, param_set)

test/test_adrv9009_p.py:231:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
test/dma_tests.py:303: in cw_loopback
    np.argmax(np.abs(np.array(getattr(sdr, p)) - np.array(param_set[p])))
adi/adrv9009.py:139: in calibrate
    return self._get_iio_dev_attr("calibrate")
adi/attribute.py:171: in _get_iio_dev_attr
    return get_numbers(self._get_iio_dev_attr_str(attr_name, _ctrl))
adi/attribute.py:159: in _get_iio_dev_attr_str
    return self._ctrl.attrs[attr_name].value
../../.local/lib/python3.8/site-packages/iio.py:706: in <lambda>
    lambda self: self._read(),
../../.local/lib/python3.8/site-packages/iio.py:762: in _read
    _d_read_attr(self._device, self._name_ascii, buf, len(buf))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

result = -5, func = <_FuncPtr object at 0x7f7289848940>, arguments = (<iio.LP__Device object at 0x7f728208fbc0>, b'calibrate', <ctypes.c_char_Array_1024 object at 0x7f7282291f40>, 1024)

    def _check_negative(result, func, arguments):
        if result >= 0:
            return result
>       raise OSError(-result, _strerror(-result))
E       OSError: [Errno 5] Input/output error

../../.local/lib/python3.8/site-packages/iio.py:70: OSError
```

I'm going to dig deeper into this. Most probably I've messed up some python installations in ubuntu. 

Cheers!
Filip

flpm2, 2020-12-17:
Managed to get past the OSError by commenting out the "calibrate" attribute step, however for some reason I see that I have 
```
________________________________ test_adrv9009_iq_loopback[param_set2-3-adi.adrv9009] ________________________________

test_iq_loopback = <function cw_loopback at 0x7ff6593759d0>, iio_uri = 'ip:analog.local', classname = 'adi.adrv9009'
channel = 3
param_set = {'gain_control_mode_chan0': 'slow_attack', 'gain_control_mode_chan1': 'slow_attack', 'trx_lo': 5000000000, 'trx_lo_chip_b': 1000000000, ...}

    @pytest.mark.iio_hardware(hardware)
    @pytest.mark.parametrize("classname", [(classname)])
    @pytest.mark.parametrize("channel", [0, 1, 2, 3])
    @pytest.mark.parametrize(
        "param_set",
        [
            dict(
                trx_lo=1000000000,
                gain_control_mode_chan0="slow_attack",
                gain_control_mode_chan1="slow_attack",
                tx_hardwaregain_chan0=-20,
                tx_hardwaregain_chan1=-20,
            #    calibrate_tx_qec_en=1,
            #    calibrate=1,
            ),
            dict(
                trx_lo=3000000000,
                gain_control_mode_chan0="slow_attack",
                gain_control_mode_chan1="slow_attack",
                tx_hardwaregain_chan0=-20,
                tx_hardwaregain_chan1=-20,
            #    calibrate_tx_qec_en=1,
            #    calibrate=1,
            ),
            dict(
                trx_lo=5000000000,
                trx_lo_chip_b=1000000000,
                gain_control_mode_chan0="slow_attack",
                gain_control_mode_chan1="slow_attack",
                tx_hardwaregain_chan0=-20,
                tx_hardwaregain_chan1=-20,
            #    calibrate_tx_qec_en=1,
            #    calibrate=1,
            ),
        ],
    )
    def test_adrv9009_iq_loopback(test_iq_loopback, iio_uri, classname, channel, param_set):
>       test_iq_loopback(iio_uri, classname, channel, param_set)

test/test_adrv9009_p.py:231:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
test/dma_tests.py:309: in cw_loopback
    sdr.tx_enabled_channels = [channel]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <adi.adrv9009.adrv9009 object at 0x7ff6543d3a90>, value = [3]

    @tx_enabled_channels.setter
    def tx_enabled_channels(self, value):
        if self._complex_data:
            if max(value) > ((self._num_tx_channels) / 2 - 1):
>               raise Exception("TX mapping exceeds available channels")
E               Exception: TX mapping exceeds available channels

adi/rx_tx.py:305: Exception
================================================== warnings summary ==================================================
```
I guess it is now safe to assume that the errors I get are due to me having looped back only TX2-RX2 of the adrv9009.
```
================================================ test session starts =================================================
platform linux -- Python 3.8.6, pytest-6.2.1, py-1.9.0, pluggy-0.13.0 -- /usr/bin/python
cachedir: .pytest_cache
[... 26 lines cut ...]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_agc[0.9-0-22-param_set2-1-adi.adrv9009] PASSED           [ 52%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-0-adi.adrv9009] FAILED   [ 54%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-1-adi.adrv9009] FAILED   [ 56%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-2-adi.adrv9009] FAILED   [ 59%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set0-0.5-20-60-3-adi.adrv9009] FAILED   [ 61%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-0-adi.adrv9009] FAILED    [ 63%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-1-adi.adrv9009] FAILED    [ 65%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-2-adi.adrv9009] FAILED    [ 68%]
test/test_adrv9009_p.py::test_adrv9009_dds_gain_check_vary_power[param_set1-0.5-0-15-3-adi.adrv9009] FAILED    [ 70%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-0-adi.adrv9009] PASSED                           [ 72%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-1-adi.adrv9009] PASSED                           [ 75%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-2-adi.adrv9009] FAILED                           [ 77%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set0-3-adi.adrv9009] FAILED                           [ 79%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-0-adi.adrv9009] PASSED                           [ 81%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-1-adi.adrv9009] PASSED                           [ 84%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-2-adi.adrv9009] FAILED                           [ 86%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set1-3-adi.adrv9009] FAILED                           [ 88%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-0-adi.adrv9009] PASSED                           [ 90%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-1-adi.adrv9009] PASSED                           [ 93%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-2-adi.adrv9009] FAILED                           [ 95%]
test/test_adrv9009_p.py::test_adrv9009_iq_loopback[param_set2-3-adi.adrv9009] FAILED                           [ 97%]
test/test_adrv9009_p.py::test_adrv9009_jesd PASSED                                                             [100%]

====================================================== FAILURES ======================================================
```

flpm2, 2020-12-17:
Triggering calibrate, triggers this I/O error.
```
 python adrv9009.py
-10
-10
TRX LO 2000000000
Traceback (most recent call last):
  File "adrv9009.py", line 68, in <module>
    sdr.calibrate
  File "/usr/local/lib/python3.8/dist-packages/pyadi_iio-0.0.7-py3.8.egg/adi/adrv9009.py", line 139, in calibrate
    return self._get_iio_dev_attr("calibrate")
  File "/usr/local/lib/python3.8/dist-packages/pyadi_iio-0.0.7-py3.8.egg/adi/attribute.py", line 171, in _get_iio_dev_attr
    return get_numbers(self._get_iio_dev_attr_str(attr_name, _ctrl))
  File "/usr/local/lib/python3.8/dist-packages/pyadi_iio-0.0.7-py3.8.egg/adi/attribute.py", line 159, in _get_iio_dev_attr_str
    return self._ctrl.attrs[attr_name].value
  File "/usr/local/lib/python3.8/dist-packages/iio.py", line 706, in <lambda>
    lambda self: self._read(),
  File "/usr/local/lib/python3.8/dist-packages/iio.py", line 762, in _read
    _d_read_attr(self._device, self._name_ascii, buf, len(buf))
  File "/usr/local/lib/python3.8/dist-packages/iio.py", line 70, in _check_negative
    raise OSError(-result, _strerror(-result))
OSError: [Errno 5] Input/output error
```

tfcollins, 2020-12-17:
Well, calibrate is a write-only attribute. It should error when read from, but the tests don't do that ....

Anyway, as for the tests with channel [0, 1, 2, 3], these are meant for FMComms8 or the ADRV9009-ZU11EG board. I guess we need to split those out. They have multiple ADRV9009's onboard.

You should be fine now.

flpm2, 2020-12-21:
I think that line 304 assert attempts a read on "calibrate":
```
297     # Verify still set
298     for p in param_set.keys():
299         if isinstance(param_set[p], str):
300             assert getattr(sdr, p) == param_set[p]
301         else:
302             time.sleep(1)
303             assert:
304                 np.argmax(np.abs(np.array(getattr(sdr, p)) - np.array(param_set[p])))
305                 < 4
306             )
"test/dma_tests.py" [Modified] 749 lines --34%--                                                   256,1         36%
```
And triggers the I/O error.

tfcollins, 2020-12-22:
hmmm. I was looking at the wrong test, sorry.

This should fix that: https://github.com/analogdevicesinc/pyadi-iio/tree/fix-adrv9009-test

-Travis

flpm2, 2020-12-23:
Yep, thanks!
I think for the sake of completness you can also update line 142 and 198 of test/test_adrv9009_p.py to remove extra channels.
Very much appreciate your help!
-Filip

closed 2020-12-23 by flpm2