analogdevicesinc/pyadi-iio #163: rx_tx() class doesn't allow for disabling of all the ADC/DAC channels

wsapia-adi, 2021-01-15:
Line 93/96 tries to take the max of the input, but if the input is an empty list (all channels disabled), a ValueError is raised as you can't take the max of an empty sequence.
https://github.com/analogdevicesinc/pyadi-iio/blob/master/adi/rx_tx.py#L93

Tx shares the same issue
https://github.com/analogdevicesinc/pyadi-iio/blob/master/adi/rx_tx.py#L304

tfcollins, 2021-01-15:
I can add an empty check, but what are the use cases here?

-Travis

wsapia-adi, 2021-01-15:
The first example that comes to mind would be if you're only looking to use the ADCs, then there's no way to turn off all of the DACs and vice-versa.

Mostly though, I just feel like this shouldn't error out. Disabling the devices may not be a common use case, but it's certainly a valid operating point.

tfcollins, 2021-01-16:
In the context of libiio (forgive the pun) channel disable is a bit strange since it can be only applied after a buffer is destroyed or before one is created. In pyadi we always disable all the channels before a buffer is created and only enable the ones you want.

From the perspective of TX, if you want to stop the current buffer you would call [tx_destroy_buffer](https://analogdevicesinc.github.io/pyadi-iio/buffers/index.html#adi.rx_tx.tx.tx_destroy_buffer). However, due to how the TX driver works this would switch you into DDS mode on most FPGA based systems. This is [documented here](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-dds/axi-dac-dds-hdl).

Now it is possible to feed DACs zeros by manipulating the raw parameter. The API could look something like this:
```
dacdev = adr.<something>
dacdev.tx_enabled_channels = None
dacdev.tx()
```

-Travis

wsapia-adi, 2021-01-18:
There's no way to disable all the Rx channels? Is there no DC power benefit to doing so, or is disabling a buffer only done in software and has no effect on the actual hardware?

tfcollins, 2021-01-18:
From the perspective of buffers, you are simply controlling muxes into the DMA. This is all done within the FPGA. Powering down channels is a chip-specific feature and is usually not runtime configurable.

Clearing the buffer or not calling rx() would have the same effect as disabling all the channels.

-Travis

wsapia-adi, 2021-01-18:
Ok, thanks for the info

PR #164 (2021-01-19): Empty enable chan
# Description

Fixes #163 and adds more features related to buffers.

## Type of change

Please delete options that are not relevant.

- [X] Bug fix (non-breaking change which fixes an issue)
- [X] New feature (non-breaking change which adds functionality)
- [X] This change requires a documentation update

# How has this been tested?

New tests have been added to validate new zeroing buffer features which are part of PR.

**Test Configuration**:
* Hardware: PlutoSDR
* OS: Ubuntu 18.04

# Documentation

Doc has been updated in regards to the new behavior.

# Checklist:

- [x]  [... cut]
closed 2021-01-19 by tfcollins