analogdevicesinc/pyadi-iio #227: OSError when pushing the buffer using the latest pyadi-iio and libiio python bindings

xwkang2019, 2021-09-08:
Dear Developers,

We were recently revisiting some scripts that we had working with ADALM-Pluto one year and half ago. However, after installing the latest version of PyADI-IIO following the tutorials on https://wiki.analog.com/resources/tools-software/linux-software/pyadi-iio. We were getting OS error when pushing a buffer. The error would occur even if we are simply using the example code provided on: https://github.com/analogdevicesinc/pyadi-iio/blob/master/examples/pluto.py

We are not sure what may be causing this and attached below is a screenshot of the error message:
[image]

It is worth noting that we have the libiio, libm2k already installed in the system (essentially have our code running normally with ADALM-2000). I then followed the guideline on https://wiki.analog.com/resources/tools-software/linux-software/pyadi-iio to install the libiio(via installer), pylibiio(via pip) and pyadi-iio(via pip). My recollection was that one year and half ago we had the pyadi working with pluto while also having the libm2k package installed. So I suspect there's any conflict occurring here.

I'll also reinstall the Windows OS and go through https://wiki.analog.com/resources/tools-software/linux-software/pyadi-iio again and see if that works.

Thanks in advance for any insights you could offer regarding the OS error issue!

tfcollins, 2021-09-09:
The first error is actually a known issue and was fixed earlier today. Grab pylibiio v0.23.1 and it will go away.

The second one looks like you are missing libiio itself or need to reinstall it.

-Travis

xwkang2019, 2021-09-13:
Thanks! The problem is now fixed!

closed 2021-09-13 by tfcollins