analogdevicesinc/pyadi-iio #315: Segmentation fault whenever trying to transmit with PlutoSDR

marsmathis, 2022-03-22:
My code:

```
Python 3.10.2 (main, Jan 15 2022, 19:56:27) [GCC 11.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adi
>>> sdr = adi.Pluto()
>>> sdr.rx()
array([ 38. +63.j, -18. -49.j, -32.-117.j, ..., -29. -12.j,  13. -52.j,
       -37. -25.j])
>>> sdr.tx([1])
[1]    54773 segmentation fault (core dumped)  python
```

I always get that segfault whenever I try to invoke `sdr.tx()`, no matter the payload. Even with the example code, this happens every time. 
The strange thing is though that I do see a very short blip on my receiver at the specified frequency when I run the following code even though the segfault gets thrown.

```
import numpy as np
import adi

sample_rate = 1e6
center_freq = 915e6

sdr = adi.ad936x.Pluto('ip:192.168.2.1')
sdr.tx_enabled_channels = [0]
sdr.sample_rate = int(sample_rate)
sdr.tx_rf_bandwidth = int(sample_rate)
sdr.tx_lo = int(center_freq)
sdr.tx_hardwaregain_chan0 = -50

N = 10000
t = np.arange(N)/sample_rate
samples = 0.5*np.exp(2.0j*np.pi*100e3*t)
# print(samples[0:100])
samples *= 2**14

sdr.tx_cyclic_buffer = True
sdr.tx(samples)
```

Any idea why this could be? Is this a bug in the module?

tfcollins, 2022-03-22:
What version of pyadi-iio are you using?

theproxy, 2022-09-07:
I am seeing this in Python 3.8.10 with pylibiio-0.23.1. Found this posting today after looking for a solution.

tfcollins, 2022-09-08:
@theproxy please create a separate issue and include your version of all packages and libiio c library used.

-Travis