analogdevicesinc/pyadi-iio #361: Trouble connecting to Pluto on Apple silicon

kingsquirrel152, 2022-08-11:
Greetings, apologies if this is the wrong forum for a question like this. I am stumped. Any insight or direction would be appreciated. 

In the simplest form I am doing something simple like:

```python
sdr = adi.ad9361(uri='ip:192.168.3.1')
sdr.sample_rate = int(sample_rate_hz)
```

But the call to set the sample_rate blows up with a key error for the channel that is being used in the library.  I can do a `iio_addr` dump and I see sampling frequency and all the usual suspects. So I think iiolib is happy. Not quite sure what abstraction level is not working correctly.

```python
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Input In [10], in <cell line: 5>()
      1 sdr._rxadc.set_kernel_buffers_count(4)
      3 sdr.rx_enabled_channels = [0, 1]
----> 5 sdr.sample_rate = int(sample_rate_hz)
      7 # Config Rx
      8 sdr.rx_lo = int(center_freq_hz)

File ~/Library/Python/3.8/lib/python/site-packages/adi/ad936x.py:188, in ad9364.sample_rate(self, rate)
    186     taps = 64
    187 # fmt: on
--> 188 current_rate = self._get_iio_attr("voltage0", "sampling_frequency", False)
    190 if self._get_iio_attr("out", "voltage_filter_fir_en", False):
    191     if current_rate <= (25000000 // 12):

File ~/Library/Python/3.8/lib/python/site-packages/adi/attribute.py:156, in attribute._get_iio_attr(self, channel_name, attr_name, output, _ctrl)
    153 def _get_iio_attr(self, channel_name, attr_name, output, _ctrl=None):
    154     """ Get channel attribute as number """
    155     return get_numbers(
--> 156         self._get_iio_attr_str(channel_name, attr_name, output, _ctrl)
    157     )

File ~/Library/Python/3.8/lib/python/site-packages/adi/attribute.py:151, in attribute._get_iio_attr_str(self, channel_name, attr_name, output, _ctrl)
    149 if not channel:
    150     raise Exception("No channel found with name: " + channel_name)
--> 151 return channel.attrs[attr_name].value

KeyError: 'sampling_frequency'
```

Note this is on Apple M2 silicon, ox 12.4.  and I've manually installed all the things via instructions here: https://pysdr.org/content/pluto.html#installing-plutosdr-driver

Thanks in advance.

tfcollins, 2022-08-11:
This issue is related to the libiio python bindings and not pyadi-iio. Unfortunately none of the staff have Apple Silicon macs so you will need to dig into the iio.py file to trace the source of the error https://github.com/analogdevicesinc/libiio/blob/master/bindings/python/iio.py I would look at the enumeration of the channel attributes.

-Travis

kingsquirrel152, 2022-08-11:
Ok. Thanks, Ill move it to Libiio.

closed 2022-08-11 by kingsquirrel152