analogdevicesinc/pyadi-iio #376: typo in ad7124.py

jbeale1, 2022-09-17:
Line 74 of this file has a typo: https://github.com/analogdevicesinc/pyadi-iio/blob/master/adi/ad7124.py
there is an extra underscore in an object member:

 for ch in self._ctrl._channels:  should actually be
 for ch in self._ctrl.channels:

having made this correction, I got the example code to work. It did not work until I made this fix. I'm using a Raspberry Pi 3B running the current Kuiper Linux distribution as of yesterday (16-Sep-2022), so I assume that setup should have worked as-is.

tfcollins, 2022-09-17:
Thanks for reporting. Looks like this is a leftover from the older bindings.

If you want to submit your change we can merge it today.

-Travis

jbeale1, 2022-09-17:
I have not actually done a pull request on github before. The only change I made to the file is that one character on line 74.

jbeale1, 2022-09-17:
I have attempted to make a pull request for this change. https://github.com/analogdevicesinc/pyadi-iio/pull/377

tfcollins, 2023-01-10:
Fixed by https://github.com/analogdevicesinc/pyadi-iio/commit/2c98af1a2dc32a2a4c73faff049a1818202415a0

closed 2023-01-10 by tfcollins