analogdevicesinc/pyadi-iio #391: AD9081 not connecting throwing OSError: [Errno 0] No error

gilv1984, 2022-12-19:
I have installed a clean copy of pyadi and tried to run the AD9081 example.
I verified I have ping to analog.local and yet I keep getting the following error:
Traceback (most recent call last):

  File "C:\ProgramData\Anaconda3\lib\site-packages\adi\context_manager.py", line 65, in __init__
    self._ctx = iio.Context(self.uri)

  File "C:\ProgramData\Anaconda3\lib\site-packages\iio.py", line 1358, in __init__
    self._context = _new_uri(_context.encode("ascii"))

  File "C:\ProgramData\Anaconda3\lib\site-packages\iio.py", line 56, in _check_null
    raise OSError(err, _strerror(err))

OSError: [Errno 0] No error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "c:\users\gil.vishinsky\ramon space\r&d comm - documents\hw\python scripts\lab_equipment\untitled0.py", line 47, in <module>
    dev = adi.ad9081("ip:analog.local")

  File "C:\ProgramData\Anaconda3\lib\site-packages\adi\ad9081.py", line 103, in __init__
    context_manager.__init__(self, uri, self._device_name)

  File "C:\ProgramData\Anaconda3\lib\site-packages\adi\context_manager.py", line 67, in __init__
    raise Exception("No device found")

Exception: No device found

a weird issue that I found is when I import iio it works fine but when looking for it in pip (pip show iio) it says it is not installed. 
I am using anaconda3 and python3.

tfcollins, 2022-12-19:
> a weird issue that I found is when I import iio it works fine but when looking for it in pip (pip show iio) it says it is not installed.
> I am using anaconda3 and python3.

iio is not the name of the dependent package. The python bindings for libiio are called pylibiio and the C library from conda-forge is called libiio-c.

> Exception: No device found

This error is likely unrelated to the python layers. Can you verify the board is booted correctly and the drivers appear. If you run this command and provide the output?

```bash
iio_attr -u ip:analog.local -d
```

gilv1984, 2022-12-20:
after several reboots of the board and a reboot to my computer I finally managed to connect (I think it was related to the ACE SW that was closed but run before and maybe kept something up.
thanks for the fast reply.
not sure if this is the place to ask but is there support for AD9082? the commands for 9081 does not work (and that makes sense because its not the same chip and they have different number of ADCs...)
thanks again for the fast reply.
gil.

tfcollins, 2022-12-20:
ACE and the prototyping tools (like pyadi-iio) have completely different software stacks. Basically, if you are using ADS9 it will only work with ACE. If you have a ZCU102, VCU118, VCU128, VCK190, ZC706, or Arria 10 they are supported by the prototyping software.  AD9082 is supported by both software solutions.

gilv1984, 2022-12-20:
Well I am in fact using the ADS9 evaluation board. 
what you are saying is that pyadi will not work with this evaluation board? 
We do possess additional VCK190 board, should I run the python script directly on the vck190 linux?

tfcollins, 2022-12-20:
Yes, the ADS9 system will only talk to ACE.

If you want to use pyadi-iio you need to use the [Kuiper SD Card](https://www.analog.com/en/design-center/evaluation-hardware-and-software/software/kuiper-linux.html)

[VCK190 doc for AD9081](https://wiki.analog.com/resources/eval/user-guides/ad9081_fmca_ebz/quickstart/versal)

-Travis

gilv1984, 2022-12-27:
Hi travis,
I found zcu111 and a zedboard. Can i use one of them to run the ad9082 with pyadi?

tfcollins, 2023-01-10:
Not the Zedboard but for the ZCU111 you would need to create a custom reference on your own for the FPGA.

tfcollins, 2023-01-20:
Closing. Reopen if still an issue

closed 2023-01-20 by tfcollins