analogdevicesinc/pyadi-iio #399: ADPD410X PyADI-IIO Examples and Setup.py

nicolechar, 2023-01-11:
Hey, I am currently facing an issue with installing PyADI-IIO for the ADPD410X series, using the user manual provided: https://wiki.analog.com/resources/eval/user-guides/circuits-from-the-lab/eval-adpd410x.
1. I have followed the whole guide step-by-step, tried installing the whole folder, installing using pip, and cloning the directory. However, all directories do not include 'setup.py'.
[image]

2. The next step indicates that there are 3 scripts for the ADPD410X example, but it is not included within the examples folder either.
[image]

I will like to understand how to acquire these two files, as I am unable to test my ADPD410X using PyADI-IIO if it cannot be installed. Please give me some advice on how to acquire these files!

Thank you!

catapangan, 2023-01-12:
Hi @nicolechar ,

Currently the scripts that the user guide is pointing to reside in the adpd410x branch of pyadi-iio (https://github.com/analogdevicesinc/pyadi-iio/tree/adpd410x)

The setup.py should be in the root folder when you clone the pyadi-iio repo.

HRosete, 2023-01-12:
Hello, 

The setup.py is already deprecated with  https://peps.python.org/pep-0632/

pyproject.toml now replaces the setup.py as well as setup.cfg. Please see this PR now merged to PYADI-IIO's master branch:  https://github.com/analogdevicesinc/pyadi-iio/pull/392

Also, please rebase your code. the README file is now updated to how to install PYADI-IIO. Thank you.

nicolechar, 2023-01-12:
> Hello,
> 
> The setup.py is already deprecated with https://peps.python.org/pep-0632/
> 
> pyproject.toml now replaces the setup.py as well as setup.cfg. Please see this PR now merged to PYADI-IIO's master branch: #392
> 
> Also, please rebase your code. The README file is now updated to how to install PYADI-IIO with pyproject.toml. Thank you.

Got it, thanks! I managed to install using `pip install pyproject.toml`. 

However, the issue with the missing folder for adpd410x_example.py still persists.
[image]
This is done following the installation of pyproject.toml.

Anyone knows where this file is and how I can install this successfully?

catapangan, 2023-01-12:
Hi @nicolechar ,

The example script is found in the adpd410x branch of pyadi-iio.

tfcollins, 2023-01-12:
@catapangan please create a PR for the example into master

catapangan, 2023-01-13:
Hi @nicolechar ,

It should be under the adpd410x folder in the examples folder: https://github.com/analogdevicesinc/pyadi-iio/tree/adpd410x/examples/adpd410x

nicolechar, 2023-01-13:
> Hi @nicolechar ,
> 
> It should be under the adpd410x folder in the examples folder: https://github.com/analogdevicesinc/pyadi-iio/tree/adpd410x/examples/adpd410x

Got it! Thanks a bunch @catapangan, I am able to install it now. (-: