analogdevicesinc/pyadi-iio #401: Error using tx() function, OS Error

Ash1295, 2023-01-13:
I am connecting to my ADRV9364Z7020 Eval board using PyADI. It is able to configure the RFIC using the example script provided and the rx() command works well but for tx(data) command it throws an OS Error: exception: access violation reading 0x0000000000000068. On probing further I understand that the len(sdr._txdac.channels) command returns 6 and refers to a set of locations but there are only 4 "altvoltage" labeled channels located in a different address from before. Please let me know if I have missed something or if there are any issues with identification of channels

tfcollins, 2023-01-13:
What example are you running?

Ash1295, 2023-01-13:
ad9361_example.py

tfcollins, 2023-01-13:
If you have the ADRV9364Z7020 you cannot use that example directly. You need to change adi.ad9361 to adi.ad9364

Ash1295, 2023-01-16:
I have changed that before running the script

tfcollins, 2023-01-17:
Can you fill in the uri, run the following, and provide the output:

```python
import adi
sdr = adi.ad9364(uri=<uri>)

for dev in sdr.ctx.devices:
    print(f"-- dev: {dev.name}")
    for chan in dev.channels:
        print(f"chan: {chan.name} {chan._id}")
```

Ash1295, 2023-01-19:
-- dev: e000b000ethernetffffffff00
chan: None temp1
-- dev: ad7291
chan: None temp0
chan: None voltage0
chan: None voltage1
chan: None voltage2
chan: None voltage3
chan: None voltage4
chan: None voltage5
chan: None voltage6
chan: None voltage7
-- dev: ad7291-bob
chan: None temp0
chan: None voltage0
chan: None voltage1
chan: None voltage2
chan: None voltage3
chan: None voltage4
chan: None voltage5
chan: None voltage6
chan: None voltage7
-- dev: ad9361-phy
chan: RX_LO altvoltage0
chan: TX_LO altvoltage1
chan: None out
chan: None temp0
chan: None voltage0
chan: None voltage0
chan: None voltage2
chan: None voltage2
chan: None voltage3
-- dev: xadc
chan: None temp0
chan: vccint voltage0
chan: vccaux voltage1
chan: vccbram voltage2
chan: vccpint voltage3
chan: vccpaux voltage4
chan: vccoddr voltage5
chan: vrefp voltage6
chan: vrefn voltage7
-- dev: cf-ad9361-dds-core-lpc
chan: TX1_I_F1 altvoltage0
chan: TX1_I_F2 altvoltage1
chan: TX1_Q_F1 altvoltage2
chan: TX1_Q_F2 altvoltage3
chan: None voltage0
chan: None voltage1
-- dev: cf-ad9361-lpc
chan: None voltage0
chan: None voltage1
-- dev: None

Ash1295, 2023-01-19:
in this only 4 tx_dac channels are showing

tfcollins, 2023-01-19:
That seems fine. Can you provide the full stack trace? Are you running locally on the board or on a remote PC?

Ash1295, 2023-01-19:
I am sorry, but I don't understand what full stack trace means.. I have connected to the board through a remote PC

tfcollins, 2023-01-19:
A stack trace is the full error from python.

Ash1295, 2023-01-20:
oh ok, thanks. Here it is:
OSError                                   Traceback (most recent call last)
Cell In[1], line 115
    113     # Send data
    114     print(indu)
--> 115     sdr.tx(iq)
    117 # Collect data
    118 for r in range(20):

File ~\AppData\Roaming\Python\Python39\site-packages\adi\rx_tx.py:513, in tx.tx(self, data_np)
    510         indx = indx + 1
    512 if not self.__txbuf:
--> 513     self.disable_dds()
    514     self._tx_buffer_size = len(data) // stride
    515     self._tx_init_channels()

File ~\AppData\Roaming\Python\Python39\site-packages\adi\dds.py:85, in dds.disable_dds(self)
     83 def disable_dds(self):
     84     """Disable all DDS channels and set all output sources to zero."""
---> 85     self.dds_enabled = np.zeros(self._num_tx_channels * 2, dtype=bool)

File ~\AppData\Roaming\Python\Python39\site-packages\adi\dds.py:125, in dds.dds_enabled(self, value)
    123 @dds_enabled.setter
    124 def dds_enabled(self, value):
--> 125     self.__update_dds("raw", value)

File ~\AppData\Roaming\Python\Python39\site-packages\adi\dds.py:49, in dds.__update_dds(self, attr, value)
     47 split_cores_indx = 0
     48 for indx in range(len(self._txdac.channels)):
---> 49     chan = self._txdac.find_channel("altvoltage" + str(indx), True)
     50     if not chan and self._split_cores:
     51         chan = self._txdac_chip_b.find_channel(
     52             "altvoltage" + str(split_cores_indx), True
     53         )

File ~\AppData\Roaming\Python\Python39\site-packages\iio.py:1199, in _DeviceOrTrigger.find_channel(self, name_or_id, is_output)
   1186 """
   1187 
   1188 Find a IIO channel by its name or ID.
   (...)
   1196     The IIO Device
   1197 """
   1198 chn = _d_find_channel(self._device, name_or_id.encode("ascii"), is_output)
-> 1199 return None if chn is None else Channel(self, chn)

File ~\AppData\Roaming\Python\Python39\site-packages\iio.py:846, in Channel.__init__(self, dev, _channel)
    840 self._channel = _channel
    841 self._dev = dev
    842 self._attrs = {
    843     name: ChannelAttr(_channel, name)
    844     for name in [
    845         _c_get_attr(_channel, x).decode("ascii")
--> 846         for x in range(0, _c_attr_count(_channel))
    847     ]
    848 }
    849 self._id = _c_get_id(self._channel).decode("ascii")
    851 name_raw = _c_get_name(self._channel)

OSError: exception: access violation reading 0x0000000000000068

tfcollins, 2023-01-20:
My guess would be that your libiio bindings might be old. Can you provide the output of:
```
pip list
```
How did you install pyadi-iio and pylibiio?

Ash1295, 2023-01-23:
I installed them through pip. Here's the list:

Package                  Version
------------------------ -----------
anyio                    3.6.2
argon2-cffi              21.3.0
argon2-cffi-bindings     21.2.0
asttokens                2.2.1
attrs                    22.2.0
Babel                    2.11.0
backcall                 0.2.0
beautifulsoup4           4.11.1
bleach                   5.0.1
certifi                  2022.12.7
cffi                     1.15.1
charset-normalizer       2.1.1
colorama                 0.4.6
comm                     0.1.2
contourpy                1.0.6
cycler                   0.11.0
debugpy                  1.6.4
decorator                5.1.1
defusedxml               0.7.1
entrypoints              0.4
executing                1.2.0
fastjsonschema           2.16.2
fonttools                4.38.0
idna                     3.4
importlib-metadata       6.0.0
ipykernel                6.19.4
ipython                  8.7.0
ipython-genutils         0.2.0
jedi                     0.18.2
Jinja2                   3.1.2
json5                    0.9.10
jsonschema               4.17.3
jupyter-client           7.4.8
jupyter-core             5.1.1
jupyter-events           0.5.0
jupyter-server           2.0.6
jupyter-server-terminals 0.4.3
jupyterlab               3.5.2
jupyterlab-pygments      0.2.2
jupyterlab-server        2.17.0
kiwisolver               1.4.4
MarkupSafe               2.1.1
matplotlib               3.6.2
matplotlib-inline        0.1.6
mistune                  2.0.4
nbclassic                0.4.8
nbclient                 0.7.2
nbconvert                7.2.7
nbformat                 5.7.1
nest-asyncio             1.5.6
notebook                 6.5.2
notebook-shim            0.2.2
numpy                    1.24.1
packaging                22.0
pandocfilters            1.5.0
parso                    0.8.3
pickleshare              0.7.5
Pillow                   9.4.0
pip                      20.2.3
platformdirs             2.6.2
prometheus-client        0.15.0
prompt-toolkit           3.0.36
psutil                   5.9.4
pure-eval                0.2.2
pyadi-iio                0.0.14
pycparser                2.21
Pygments                 2.14.0
pylibiio                 0.23.1
pyparsing                3.0.9
pyrsistent               0.19.3
python-dateutil          2.8.2
python-json-logger       2.0.4
pytz                     2022.7
pywin32                  305
pywinpty                 2.0.9
PyYAML                   6.0
pyzmq                    24.0.1
requests                 2.28.1
scipy                    1.9.3
Send2Trash               1.8.0
setuptools               49.2.1
six                      1.16.0
sniffio                  1.3.0
soupsieve                2.3.2.post1
stack-data               0.6.2
terminado                0.17.1
tinycss2                 1.2.1
tomli                    2.0.1
tornado                  6.2
traitlets                5.8.0
urllib3                  1.26.13
wcwidth                  0.2.5
webencodings             0.5.1
websocket-client         1.4.2
zipp                     3.11.0

tfcollins, 2023-01-23:
That seems fine as well. Can you try reinstalling libiio?

tfcollins, 2023-04-10:
Closing due to inactivity. Reopen if still an issue

closed 2023-04-10 by tfcollins