analogdevicesinc/pyadi-iio #461: Add example for AD9361 using modulated signal

cyber-g, 2023-08-05:
Dear Project Maintainers,
I would like to contribute to pyadi-iio,  and I am preparing an additional example for the AD9361 to achieve the transmission of modulated signals. 

The script is based on the readily available example : [ad9361_example.py](https://github.com/analogdevicesinc/pyadi-iio/blob/master/examples/ad9361_example.py) with some substantial additions (mostly to create a band-limited baseband signal) with some code lines obtained from [Filtering_Part2.ipynb](https://github.com/unpingco/Python-for-Signal-Processing/blob/master/Filtering_Part2.ipynb). 

As a result, I want to ensure that the authorship is attributed correctly to all relevant contributors and I have the following questions:
- What is the preferred method to attribute authorship for this particular Python script? 
- Should I also list myself as author ?
- If other authors should be acknowledged, what would be the appropriate way to format the authorship details within the script and any associated documentation?

Any guidance on this matter would be highly appreciated.

cyber-g, 2023-08-05:
Just in case, you can find my proposed example [here](https://github.com/cyber-g/pyadi-iio/blob/master/examples/ad9361_example_bandlimited_signal.py) before I proceed to pull request.

tfcollins, 2023-08-07:
Hi @cyber-g, since the code is unlicensed I would point you here: https://choosealicense.com/no-permission/ . If you have permission of the original author, we will license the example under ADI-BSD as noted here: https://github.com/analogdevicesinc/pyadi-iio/blob/master/CONTRIBUTING.md . Attribution should be listed as part of the copyright, similar to https://github.com/analogdevicesinc/pyadi-iio/blob/master/adi/sshfs.py#L1

cyber-g, 2023-08-21:
Thank you for your answer ; I realized that [Filtering_Part2.ipynb](https://github.com/unpingco/Python-for-Signal-Processing/blob/master/Filtering_Part2.ipynb) is part of a project under Creative Commons Attribution 3.0 Unported License (see: https://github.com/unpingco/Python-for-Signal-Processing/blob/master/LICENSE.html). 

In this case, I will give appropriate credit, provide a link to the license, and indicate changes made in the header of the file.

closed 2023-11-18 by tfcollins