analogdevicesinc/pyadi-iio #462: Attribute error for run code from example

no111u3, 2023-08-09:
Try from Pluto example (Mac OS 13.5, M2, arm64)
```
>>> import adi
>>> sdr = adi.Pluto('ip:pluto.local')
>>> sdr.rx_lo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.11/site-packages/adi/ad936x.py", line 193, in rx_lo
    return self._get_iio_attr("altvoltage0", "frequency", True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/adi/attribute.py", line 127, in _get_iio_attr
    self._get_iio_attr_str(channel_name, attr_name, output, _ctrl)
  File "/opt/homebrew/lib/python3.11/site-packages/adi/attribute.py", line 122, in _get_iio_attr_str
    return channel.attrs[attr_name].value
           ~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'frequency'
```

tfcollins, 2023-08-10:
This is a known issue with the libiio python bindings. There is a related issue here: https://github.com/analogdevicesinc/libiio/issues/871

The development team doesn't have Apple Silicon machines so think is hard for us to debug at the moment. If you can get the requested information from the issue it would be helpful.

-Travis

no111u3, 2023-08-10:
First of all, thanks for your fast answer.
For the begin I get info from iio_info for check how it works (I already use this library in sdrangel software and it works good):
```
% iio_info -u 'ip:pluto.local'
Library version: 0.23 (git tag: v0.23)
Compiled with backends: xml ip usb
IIO context created with network backend.
Backend version: 0.24 (git tag: v0.24)
Backend description string: 192.168.2.1 Linux (none) 5.10.0-98231-g9dfba10b795d #54 SMP PREEMPT Mon Jul 11 14:38:48 CEST 2022 armv7l
IIO context has 9 attributes:
	hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9364)
	hw_model_variant: 1
	hw_serial: 10447376de0b000e00001f00e99f99a2f0
	fw_version: v0.35
	ad9361-phy,xo_correction: 40000046
	ad9361-phy,model: ad9364
	local,kernel: 5.10.0-98231-g9dfba10b795d
	uri: ip:pluto.local
	ip,ip-addr: 192.168.2.1
IIO context has 4 devices:
	iio:device0: ad9361-phy
		9 channels found:
			altvoltage1: TX_LO (output)
			8 channel-specific attributes found:
				attr  0: external value: 0
				attr  1: fastlock_load value: 0
				attr  2: fastlock_recall ERROR: Invalid argument (22)
				attr  3: fastlock_save value: 0 181,181,181,181,181,181,181,181,181,181,181,181,181,181,181,181
				attr  4: fastlock_store value: 0
				attr  5: frequency value: 2450000000
				attr  6: frequency_available value: [46875001 1 6000000000]
				attr  7: powerdown value: 0
			voltage0:  (input)
			15 channel-specific attributes found:
				attr  0: bb_dc_offset_tracking_en value: 1
				attr  1: filter_fir_en value: 0
				attr  2: gain_control_mode value: slow_attack
				attr  3: gain_control_mode_available value: manual fast_attack slow_attack hybrid
				attr  4: hardwaregain value: 46.000000 dB
				attr  5: hardwaregain_available value: [-1 1 73]
				attr  6: quadrature_tracking_en value: 1
				attr  7: rf_bandwidth value: 1500000
				attr  8: rf_bandwidth_available value: [200000 1 56000000]
				attr  9: rf_dc_offset_tracking_en value: 1
				attr 10: rf_port_select value: A_BALANCED
				attr 11: rf_port_select_available value: A_BALANCED B_BALANCED C_BALANCED A_N A_P B_N B_P C_N C_P TX_MONITOR1 TX_MONITOR2 TX_MONITOR1_2
				attr 12: rssi value: 68.00 dB
				attr 13: sampling_frequency value: 4000000
				attr 14: sampling_frequency_available value: [2083333 1 61440000]
			voltage3:  (output)
			8 channel-specific attributes found:
				attr  0: filter_fir_en value: 0
				attr  1: raw value: 306
				attr  2: rf_bandwidth value: 18000000
				attr  3: rf_bandwidth_available value: [200000 1 40000000]
				attr  4: rf_port_select_available value: A B
				attr  5: sampling_frequency value: 4000000
				attr  6: sampling_frequency_available value: [2083333 1 61440000]
				attr  7: scale value: 1.000000
			altvoltage0: RX_LO (output)
			8 channel-specific attributes found:
				attr  0: external value: 0
				attr  1: fastlock_load value: 0
				attr  2: fastlock_recall ERROR: Invalid argument (22)
				attr  3: fastlock_save value: 0 59,59,59,59,59,59,59,59,59,59,59,59,59,59,59,59
				attr  4: fastlock_store value: 0
				attr  5: frequency value: 92400000
				attr  6: frequency_available value: [70000000 1 6000000000]
				attr  7: powerdown value: 0
			voltage2:  (output)
[... 176 lines cut ...]
				attr  3: label ERROR: Socket operation on non-socket (38)
				attr  4: samples_pps ERROR: Operation not supported by device (19)
				attr  5: sampling_frequency value: 4000000
				attr  6: sampling_frequency_available value: 4000000 500000
			voltage1:  (input, index: 1, format: le:S12/16>>0)
			7 channel-specific attributes found:
				attr  0: calibbias value: 0
				attr  1: calibphase value: 0.000000
				attr  2: calibscale value: 1.000000
				attr  3: label ERROR: Socket operation on non-socket (38)
				attr  4: samples_pps ERROR: Operation not supported by device (19)
				attr  5: sampling_frequency value: 4000000
				attr  6: sampling_frequency_available value: 4000000 500000
		2 device-specific attributes found:
				attr  0: sync_start_enable value: disarm
				attr  1: sync_start_enable_available value: arm
		2 buffer-specific attributes found:
				attr  0: data_available value: 0
				attr  1: length_align_bytes value: 8
		2 debug attributes found:
				debug attr  0: pseudorandom_err_check value: CH0 : PN9 : Out of Sync : PN Error
CH1 : PN9 : Out of Sync : PN Error
				debug attr  1: direct_reg_access value: 0x0
		No trigger on this device
```

no111u3, 2023-08-10:
Than I try to reach out by `iio_attr` it also accessibly:
```
% iio_attr -u 'ip:pluto.local' -c 'ad9361-phy' 'altvoltage0' 'frequency'
2399999998
```
But in python I reach out another result:
```python3
>>> import iio
>>> ctx = iio.Context('ip:pluto.local')
>>> ctx.devices[0].name
'ad9361-phy'
>>> ctx.devices[0].channels[0].name
'RX_LO'
>>> ctx.devices[0].channels[0].attrs
{'external': <iio.ChannelAttr object at 0x100e17e50>}
```

no111u3, 2023-08-10:
Also I suppose that only one attribute for each channel got:
```
>>> for dev in ctx.devices:
...     for chan in dev.channels:
...             attrs_count = len(chan.attrs)
...             print(f'{dev.name}: {chan.name}: {attrs_count}')
...
ad9361-phy: RX_LO: 1
ad9361-phy: TX_LO: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
xadc: None: 1
xadc: vccint: 1
xadc: vccaux: 1
xadc: vccbram: 1
xadc: vccpint: 1
xadc: vccpaux: 1
xadc: vccoddr: 1
xadc: vrefp: 1
xadc: vrefn: 1
xadc: None: 1
cf-ad9361-dds-core-lpc: TX1_I_F1: 1
cf-ad9361-dds-core-lpc: TX1_I_F2: 1
cf-ad9361-dds-core-lpc: TX1_Q_F1: 1
cf-ad9361-dds-core-lpc: TX1_Q_F2: 1
cf-ad9361-dds-core-lpc: None: 1
cf-ad9361-dds-core-lpc: None: 1
cf-ad9361-lpc: None: 1
cf-ad9361-lpc: None: 1
```

no111u3, 2023-08-10:
I added debug to channel constructor:
```python3
print([
                _c_get_attr(_channel, x).decode("ascii")
                for x in range(0, _c_attr_count(_channel))
            ])
```
And got strange output:
```python3
['external', 'external', 'external', 'external', 'external', 'external', 'external', 'external']
['bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en']
['filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en']
['external', 'external', 'external', 'external', 'external', 'external', 'external', 'external']
['filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en']
['input']
['filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en', 'filter_fir_en']
['bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en', 'bb_dc_offset_tracking_en']
['voltage_filter_fir_en']
ad9361-phy: RX_LO: 1
ad9361-phy: TX_LO: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
ad9361-phy: None: 1
```

no111u3, 2023-08-10:
This issue generates by wrong bindings signatures fixed by https://github.com/analogdevicesinc/libiio/commit/1246cb031728f4958cfb4324678f600e637bf21c

no111u3, 2023-08-10:
@tfcollins please look at my investigation, I think I found the main issue and possible solution.

tfcollins, 2023-08-10:
If you use the iio.py from the libiio master branch does it resolve the issue?

no111u3, 2023-08-10:
Yes sure, it fully resolve my issue

tfcollins, 2023-08-10:
kk thanks for checking. I'll be creating the python release soon for v0.25 and we'll update the pyadi-iio dependencies

PR #463 (2023-08-10): Update pylibiio dep to fix #462
# Description

Update pylibiio dep to fix issue with Apple Silicon.

Fixes #462 

## Type of change

Please delete options that are not relevant.

- [X] Bug fix (non-breaking change which fixes an issue)

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] I have signed off all commits and they contain "Signed-off by: <insert name>"
- [x] I have made corresponding changes to the documentation
- [x] My changes gen [... cut]
commit 3007fc08 (2023-08-10): Update pylibiio dep to fix #462
commit e89dc876 (2023-08-10): Merge pull request #463 from analogdevicesinc/tfcollins/update-libiio

Update pylibiio dep to fix #462
closed 2023-08-10 by tfcollins