analogdevicesinc/pyadi-iio #468: AD9361 Wide Bandwidth Noise Floor Shape

cyber-g, 2023-08-22:
Hi,
I am having unexpected shape of the noise floor when using the RX AD9361 with 61.44MSps. 

I use the [ad9361_example.py](https://github.com/analogdevicesinc/pyadi-iio/blob/master/examples/ad9361_example.py) in which I just set [L17](https://github.com/analogdevicesinc/pyadi-iio/blob/master/examples/ad9361_example.py#L17) : 
```
sdr.sample_rate = int(61.44e6)
```

The displayed spectrum is as shown here : 
[image]

The noise close to 0 Hz is not the issue for now, the one which is wrong for me is the noise regrowth is the range of 15MHz < |f| <30 MHz. 

The issue is also being discussed at ADI Engineer Zone ([AD9361 Wide Bandwidth Noise Floor Shape with PyADI-IIO - Q&A - Design Support AD9361/AD9363/AD9364 - EngineerZone](https://ez.analog.com/rf/wide-band-rf-transceivers/design-support/f/q-a/572292/ad9361-wide-bandwidth-noise-floor-shape-with-pyadi-iio)) but I feel that, this is more a configuration issue which should be discussed in the PyADI-IIO project (it is not a hardware design issue).

tfcollins, 2023-08-22:
The noise shaping here will be based on 3 things:
1. The actual shape of the sigma delta
2. The magnitude compensation filter, aka the programmable FIR
3. The bandwidth of the analog filter

If you have an undesirable response for a specific sample rate, increase (3) and/or design a custom FIR with this tool: https://wiki.analog.com/resources/eval/user-guides/ad-fmcomms2-ebz/software/filters

cyber-g, 2023-08-22:
Thank you for your clear explanation, it helped me to understand critical elements and to solve my issue. 

After spending a while analyzing the [ad936x-filter-wizard](https://github.com/analogdevicesinc/ad936x-filter-wizard), I figured out that we can easily generate the filter coefficients with few code lines. 
I put them here, so that you/others may reproduce the results: 
```matlab
% Create a partial structure for the filter design: 
fdp.Rdata   = 61.44e6;  % Output data rate in Hz
fdp.Fpass   = 56e6/2;   % FIR passband edge in Hz
fdp.Fstop   = 58e6/2;   % FIR stopband edge in Hz
fdp.Fcutoff = 56e6;     % Cutoff frequency in Hz
fdp.Astop   = 30;       % Stopband attenuation in dB

% Complete the filter design structure
fdp = cook_input(fdp)

% Perform the filter design
fir_struct = design_filter(fdp)

% Plot the filter response
fvtool(fir_struct.firtaps, 'Fs', fdp.Rdata)
```
In this case, the response of the filter is as shown in the following plot. 
[image]
which exhibits the "unexpected" out-of-band regrowth. 

Furthermore, during the code analysis, I realized I could also plot the frequency response of filters exported as `.ftr` files. Then, recalling a related question I asked few month ago ([Missing documentation for 61_44_28MHz.ftr](https://github.com/analogdevicesinc/iio-oscilloscope/issues/390)), I decided to plot the frequency response of [61_44_28MHz.ftr](https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/filters/61_44_28MHz.ftr). 
[image]
which also exhibits the "unexpected" out-of-band regrowth. 

Finally, I realized that `Fpass` and `Fstop` is not properly set for my purpose of a simple wideband receiver. 
When I fix them to : 
```matlab
fdp.Fpass   = 56e6;   % FIR passband edge in Hz
fdp.Fstop   = 58e6;   % FIR stopband edge in Hz
```
I get the following frequency response: 
[image]
I still need to test this filter to check if the inband regrowth is properly compensating the previous filter stages or if it creates additional distortions.

cyber-g, 2023-08-22:
I have one remaining question related to this custom filter design procedure: 
what is the appropriate way to export the `fir_struct` obtained with `design_filter()` programmatically from Matlab so that the generated filter configuration can be easily loaded in PyADI-IIO (which I expect to be done with `sdr.filter = "custom_filter.ftr"`)

tfcollins, 2023-08-23:
There are no functions to do this programmatically. That code is implemented within the GUI itself.

cyber-g, 2023-08-23:
Thanks for the confirmation. 

To test the filter, I decided to use `iio-oscilloscope` available on the OS booted by the Zynq card. 

By manipulating the GUI, I then (finally!... :man_facepalming: ) understood that the signal produced by `x = sdr.rx()` in the example `ad9361_example.py` is actually the output **without FIR processing**! and that the noise spectral regrowth is, in this case, mostly (if not only) related to the sigma-delta noise. 

For my memory and maybe for other possible interested person, the process of my investigation is summarized in the following pdf which shows different configurations using `iio-oscilloscope`: 
[IIO-OSC_AD936x_QAM16_FIR_investigations.pdf](https://github.com/analogdevicesinc/pyadi-iio/files/12423625/IIO-OSC_AD936x_QAM16_FIR_enable_disable.pdf)

I finally reached a "flat" noise using the [LTE20_MHz.ftr](https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/filters/LTE20_MHz.ftr). Enforcing the `Sample Rate` and `RF Bandwidth` after FIR configuration provides me a satisfying spectrum but I still need to perform further tests to validate the configuration in particular, in the case of a signal occupying 50MHz. In case of success, I plan to make a pull request to provide a "clean" full wideband TX/RX transmission.

closed 2023-11-17 by tfcollins