analogdevicesinc/pyadi-iio #500: Adalm pluto dual receive not working despite modifying fw_setenv

Argonghost, 2023-11-24:
Hi, ,, im trying to use dual receive on ADALM pluto and i followed all the steps of changing fw_set_env parameters to be AD9361...

then, below is the code im trying to run, followed by the error i keep getting:

`	
import adi
import matplotlib.pyplot as plt
import numpy as np
 
# Create radio
sdr = adi.ad9361("ip:192.168.2.1")
samp_rate = 20e6
 
'''Configure Rx properties'''
sdr.rx_enabled_channels = [0, 1]
sdr.sample_rate = int(samp_rate)
sdr.rx_rf_bandwidth = int(2e6)
sdr.rx_lo = int(2.71e9)
sdr.rx_buffer_size = 8192
sdr.gain_control_mode_chan0 = "manual"
sdr.gain_control_mode_chan1 = "manual"
sdr.rx_hardwaregain_chan0 = int(60)
sdr.rx_hardwaregain_chan1 = int(60)
samples = sdr.rx() # receive samples off Pluto
while 1:
    x = sdr.rx()
    a = np.real(x[0][:])
    b = np.real(x[1][:])
    corr = np.correlate(np.real(a), np.real(b), mode='full')
    plt.clf()
    plt.plot(corr)
    # plt.ylim([-20e6, 20e6])
    plt.draw()
    plt.pause(0.05)
    print(np.argmax(corr))
plt.show()`

`Traceback (most recent call last):
  File "/home/argonghost/Desktop/new.py", line 17, in <module>
    sdr.gain_control_mode_chan1 = "manual"
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/ad936x.py", line 223, in gain_control_mode_chan1
    self._set_iio_attr("voltage1", "gain_control_mode", False, value)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/attribute.py", line 71, in _set_iio_attr
    raise ex
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/attribute.py", line 69, in _set_iio_attr
    channel.attrs[attr_name].value = str(value)
AttributeError: 'NoneType' object has no attribute 'attrs'
``

Any help as to why im unable to get the dual receive to work

tfcollins, 2023-11-25:
Can you reboot Pluto and print out the fw variables you set and paste the output here?

Argonghost, 2023-11-26:
@tfcollins Issue resolved.  the firmware update i did was done incorrectly .

Thanks

closed 2023-11-26 by Argonghost