analogdevicesinc/pyadi-iio #506: how to connect a buffer rx() to the ADC?

karu2003, 2023-12-10:
how to connect a buffer rx() to the ADC?

I did it by analogy with other ADCs, but it doesn’t work.

    target_device = "ad7476a"
    samples = 100
    dev = ctx.find_device(target_device)
    buf = iio.Buffer(dev, samples)

or add   to  class  __init__
    rx.__init__(self)

     ADC = ad7476.ad7476a("ip:analog.local")
     ADC.rx_buffer_size = samples
     ADC.rx_enabled_channels = [ad_channel]
     data = ADC.rx()

I get an error

analog@analog:~/rpi_ad7476 $ /usr/bin/python3 /home/analog/rpi_ad7476/ad7476.py
Traceback (most recent call last):
  File "/home/analog/rpi_ad7476/ad7476.py", line 65, in <module>
    buf = iio.Buffer(dev, samples)
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 981, in __init__
    self._buffer = _create_buffer(device._device, samples_count, cyclic)
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 56, in _check_null
    raise OSError(err, _strerror(err))
OSError: [Errno 22] Invalid argument

<<<<<<<<<<<<<<<<<<<<<<<<<

Traceback (most recent call last):
  File "/home/analog/rpi_ad7476/ad7476.py", line 63, in <module>
    data = ADC.rx()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 295, in rx
    data = self.__rx_non_complex()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 268, in __rx_non_complex
    x = self.__rx_buffered_data()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 234, in __rx_buffered_data
    self._rx_init_channels()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 197, in _rx_init_channels
    self.__rxbuf = iio.Buffer(self._rxadc, self.__rx_buffer_size, False)
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 981, in __init__
    self._buffer = _create_buffer(device._device, samples_count, cyclic)
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 56, in _check_null
    raise OSError(err, _strerror(err))
OSError: [Errno 22] Invalid argument

tfcollins, 2023-12-10:
You need to enable valid channels in both cases before creating buffers. In the pyadi example you are enabling channel 1 which does not exist. Try enabling channel 0

karu2003, 2023-12-11:
The problem is not the channel number. To test them, I have two channels and the voltage is on channel 1.
I switched to version 0.25.
There was a problem with permission denied.
i need to find a way to make this accessible to all users.

I installed the trigger manuel.
    modprobe iio_trig_sysfs
    cd /sys/bus/iio/devices/iio_sysfs_trigger
    echo 123 > add_trigger
    cd /sys/bus/iio/devices/trigger0
    cat name # should give sysfstrig123
    cd /sys/bus/iio/devices/iio:device0
    echo sysfstrig123 > trigger/current_trigger
    echo 1 > scan_elements/in_voltage0_en
    echo 1 > buffer/enable

Now i have two problems. "High-speed mode not enabled" and "Connection timed out"

analog@analog:~/ad7476_025 $ /usr/bin/python3 /home/analog/ad7476_025/ad7476.py
WARNING: High-speed mode not enabled
Traceback (most recent call last):
  File "/home/analog/ad7476_025/ad7476.py", line 99, in <module>
    data = ADC.rx()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 295, in rx
    data = self.__rx_non_complex()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 268, in __rx_non_complex
    x = self.__rx_buffered_data()
  File "/home/analog/.local/lib/python3.9/site-packages/adi/rx_tx.py", line 235, in __rx_buffered_data
    self.__rxbuf.refill()
  File "/home/analog/.local/lib/python3.9/site-packages/iio.py", line 1003, in refill
    _buffer_refill(self._buffer)
  File "/home/analog/.local/lib/python3.9/site-packages/iio.py", line 62, in _check_negative
    raise OSError(-result, _strerror(-result))
TimeoutError: [Errno 110] Connection timed out

tfcollins, 2023-12-13:
Before you changed to the dynamic changes here https://github.com/karu2003/rpi_ad7476/commit/0eabfdef687148024c50cb186f4083f920d9b728 you were indexing into an undefined channel. Be aware with dynamic channels there is no guarantee of channel ordering.

For the other two issues, your device does not support high speed mode so you can do anything. Please ignore the warning. As for the timeout, you likely need to configure a trigger for the device during the init

karu2003, 2023-12-15:
Before christmas I figured out how to use a trigger and a buffer. :) 
https://github.com/karu2003/rpi_ad7476

closed 2023-12-15 by karu2003