analogdevicesinc/pyadi-iio #508: hrtimer in pyadi?

karu2003, 2023-12-16:
how to work with hrtimer in pyiio?

tfcollins, 2023-12-19:
You would need to enable the RTC trigger https://wiki.analog.com/software/linux/docs/iio/iio-trig-sysfs#adding_linux_driver_support and try to map it to your device first

karu2003, 2023-12-22:
Is there no support in PyAdi-iio.

tfcollins, 2023-12-22:
The HR timer is no different than any other trigger. As long as you can verify you can use it to drive/trigger your device, you simply set it as a source in your init like so: https://github.com/analogdevicesinc/pyadi-iio/blob/15b7cb999cb5031dfc4f7c8dfa1801a418926fe5/adi/adis16507.py#L34

karu2003, 2023-12-23:
Thank you. I tried this right away. I made my own timer.
https://github.com/karu2003/rpi_ad7476/blob/master/hrtimer.py

tfcollins, 2024-01-10:
I think this can be closed now

closed 2024-01-10 by tfcollins