analogdevicesinc/pyadi-iio #521: "Exception: Channel voltage2 not found" in ad9361 example

ea-chang, 2024-01-24:
Hi, I am having the same issue. I have the latest ADALM-PLUTO and I am using the ad9361 example (where they do sdr.rx_enabled_channels = [0], sdr.tx_enabled_channels = [0]) but I am getting the same "Exception: Channel voltage2 not found". These are my settings. What should I check/change?

[image]

[image]

_Originally posted by @ea-chang in https://github.com/analogdevicesinc/pyadi-iio/issues/445#issuecomment-1908964933_

tfcollins, 2024-01-25:
If you are using 1r1t mode use the Pluto or ad9364 class.

tfcollins, 2024-02-05:
Closing due to inactivity.

closed 2024-02-05 by tfcollins