analogdevicesinc/pyadi-iio #529: AttributeError: module 'adi' has no attribute 'adpd410x'

GoldenxSun, 2024-02-09:
I installed pyadi-iio as shown in the user guide "EVAL-ADPD410x-ARDZ User Guide" (https://wiki.analog.com/resources/eval/user-guides/circuits-from-the-lab/eval-adpd410x) but when I try the first example script (py adpd410x_example.py) I always get this error message "AttributeError: module 'adi' has no attribute 'adpd410x'" even if the library adi is installed in site-packages with all files requiered.  

[image]

I also already libiio and iio oscilloscope to verify that the hardware is running.
Is there anything I forgot to install or include?

tfcollins, 2024-02-09:
Can you print the library version?

python -c "import adi ; print(adi.__version__)"

GoldenxSun, 2024-02-09:
Yes of course. It will throw the same error: "AttributeError: module 'adi' has no attribute '__version__'" as in the following picture

[image]

tfcollins, 2024-02-09:
Are you sure you have the module installed?  Can you provide the output of pip list and inspect adi:
```
pip list
```

```
python -c "import adi ; print(dir(adi))"
```

GoldenxSun, 2024-02-10:
Yes pip list shows I have adi.init installed. Seems like I haven't installed the right module.

[image]

and dir(adi) print the following output:

[image]

GoldenxSun, 2024-02-13:
I installed it via pypi by entering the command "pip install pyadi-iio". It then installed all the libraries in the python site-packages folder. Now when I run the command again it says all required libraries are already installed: "requirement already satisfied"

GoldenxSun, 2024-02-13:
That's why I am confused. It tells me requirements are already satisfied but on the other hand it can't find the module. Even if it's fully installed in the python site-packages. I have already gone through all the recommended steps and even uninstalled and reinstalled the library.

tfcollins, 2024-02-13:
Have you removed adi.util? If not then its likely they is higher priority on the import list and will trump pyadi-iio.

GoldenxSun, 2024-02-13:
Where can I find adi.util?

tfcollins, 2024-02-13:
Its the package you installed through pip. So running this will remove it:
```
pip uninstall adi.init
```

GoldenxSun, 2024-02-13:
Okay thank you. But now I get a different error **TypeError: argument of type 'NoneType' is not iterable** after running the following command:

`python -c "import adi ; print(adi.__version__)"`

[image]

tfcollins, 2024-02-13:
You are missing the libiio dependency. See here https://analogdevicesinc.github.io/pyadi-iio/guides/quick.html

GoldenxSun, 2024-02-14:
Okay, thanks, that worked! Sorry for the misunderstanding. Also, I had to adjust the files cn0540.py, adis16507.py, adxl345.py, adxrs290.py and ltc2983.py because np.float was deprecated and to use the numpy scalar type specifically, I had to use np.float64.

[image]

tfcollins, 2024-02-14:
Can you provide your installed dependency list and python version?
```
pip list
```
``
python --version
```

GoldenxSun, 2024-02-14:
Yes of course

[image]

tfcollins, 2024-02-14:
> Okay, thanks, that worked! Sorry for the misunderstanding. Also, I had to adjust the files cn0540.py, adis16507.py, adxl345.py, adxrs290.py and ltc2983.py because np.float was deprecated and to use the numpy scalar type specifically, I had to use np.float64.

I'm a bit confused here. Do you have a really old clone of the pyadi-iio repo you are using?  This was fixed a long time ago https://github.com/analogdevicesinc/pyadi-iio/commit/ef365218a2b3e536ed734d3645a813cb64bc699d   If you are working in a source tree it should take precedence over installed packages with the same name.

GoldenxSun, 2024-02-14:
I understand. I was working in the apdp410x branch and probably used an old version of pyadi-iio because of that. This was because I was working with the EVAL-ADPD410x-ARDZ and the example programs are only available under the adpd410x branch. The commit https://github.com/analogdevicesinc/pyadi-iio/commit/ef365218a2b3e536ed734d3645a813cb64bc699d is newer than the apdp410x branch.

tfcollins, 2024-02-14:
ok. Please don't use branches besides main or packages from pypi/conda. If there is doc pointing to other branches let me know

closed 2024-02-14 by tfcollins