analogdevicesinc/pyadi-iio #535: Possible to use adrv9009_zu11eg_multi with disparate fmcomms8

cafeclimber, 2024-02-26:
Hi, I was wondering if it was possible to use the adrv9009_zu11eg_multi class but with some devices with the FMCOMMS8 present and some without?

Setting fmcomms8=False when constructing, everything works, but it would be cool if it were possible to, in my case, go from 32-28 channels instead of having to go 32->16

Thanks!

tfcollins, 2024-02-26:
Do you have a separate card in mind? Do you have requirements on synchronization performance?

cafeclimber, 2024-02-27:
So in the short term, the set up would be 

3 adrv9009-zu11eg on FMCs with FMCOMMS8s attached

And 1 without an FMCOMMS8. 

I wasn't sure if there was any way to use these together with the `multi` class to get 28 channels.

tfcollins, 2024-03-11:
I added an enhancement here to support this type of configuration https://github.com/analogdevicesinc/pyadi-iio/tree/tfcollins/zu11eg-fm8-enhancement I can't test this myself so if you have any feedback it would be helpful