analogdevicesinc/pyadi-iio #544: Wrong SI reading for ad7124-8

boweiliu0428, 2024-03-12:
https://github.com/analogdevicesinc/pyadi-iio/blob/main/adi/rx_tx.py#L206 says 
`x[i][samp] = (raw + rx_offset[i]) * rx_scale[i]` 
but for ad7124-8 it should be
`x[i][samp] = (raw) * rx_scale[i] + rx_offset[i]` 
according to https://wiki.analog.com/resources/tools-software/linux-drivers/iio-adc/ad7124

This has resulted in wrong readings in "SI" units (mV) when running ad7124-8 in bipolar mode.

tfcollins, 2024-03-13:
Hi,

So iio/libiio standard equation is
```
x[i][samp] = (raw + rx_offset[i]) * rx_scale[i]
```

Can you please verify with the forum here that the driver is correct https://ez.analog.com/linux-software-drivers/f/q-a

boweiliu0428, 2024-03-29:
You are right. I was using the latest released version 0.0.16 in pip. Now I compiled 0.0.17 from source and everything works fine, thanks to your [change](https://github.com/analogdevicesinc/pyadi-iio/pull/507). Please release 0.0.17 into pip soon.

closed 2024-03-29 by boweiliu0428