analogdevicesinc/pyadi-iio #547: SPI interfaceing with ad5544 and ad5592r py and c/c++

JohnMunnie, 2024-03-27:
I am look to test the ad5544 and ad5592 toward circuit design how ever there is are no available library or example test guides avaible.

the ad5592r has some examples but no schematic to follow, as well as using a variable atte in the statement

for attr in dir(my_ad5592r):

in the (pyadi-110 examples/ad5592r_example.py ) example, with no reference to its initialisation.

would you be able to provide a brief guide on the interfaceing of the AD5544 and AD5592

tfcollins, 2024-03-29:
I'm not really sure exactly what you are looking for. The pyadi-iio interfaces use in-kernel IIO drivers the manage the devices themselves, which would include initialization. Are you familiar with the driver https://wiki.analog.com/resources/tools-software/linux-drivers/iio-dac/ad5592r ?

-Travis

JohnMunnie, 2024-04-10:
Hi, Travis.

Thank you for the above,  its another useful option.

I tried to install the pyadi-iio abstraction module but met with PACKAGE DOES NOT MATCH WITH HASHES FROM REQUIREMENTS FILE.

I believe its just a case of option overload.

with the 5544 however there seems to be no files available to hand.

tfcollins, 2024-04-12:
The driver is required to use the python layer provided by pyadi 

> I believe its just a case of option overload.

I don't understand what you mean

JohnMunnie, 2024-04-12:
I have been trying to install the driver. pip install ptadi-iio eventually work but the runng the example codes returns a numpy import error and a link to resolve the error that does not work.

option overload because i am looking a .h and c file and require .py or a guide of some sort.

In some cases there is direction on overlays but cloning the repository to do this always comes with a error approximatly  70%  into the update or install should i say.

it is a rather strenuos task to find the ideal process for this.

where the datasheet has useful information on operation, setting up is a task in itself and then understanding what the examples are doing.

tfcollins, 2024-04-18:
If you have questions about the drivers or setting up the drivers you will need to get support here https://ez.analog.com/linux-software-drivers/ We can only address pyadi-iio specific question in this forum.

closed 2024-04-18 by tfcollins