analogdevicesinc/pyadi-iio #555: ADRV9002 carrier frequency property -- wrong setters/getters

michaelkimmer, 2024-04-11:
In ./adi/adrv9002.py (lines 687-721) there are wrong attributes names in setters/getters of the carrier frequencies:

------------------------------------

Input:
sdr = adi.adrv9002(uri="ip:analog.local")
sdr.tx0_lo = 2400e6

Output:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\micha\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\adi\adrv9002.py", line 708, in tx0_lo
    return self._get_iio_attr("altvoltage2", "frequency", True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\adi\attribute.py", line 127, in _get_iio_attr
    self._get_iio_attr_str(channel_name, attr_name, output, _ctrl)
  File "C:\Users\micha\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\adi\attribute.py", line 122, in _get_iio_attr_str
    return channel.attrs[attr_name].value
           ~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'frequency'

-----------------------------------------

I found errors in ./adi/adrv9002.py on lines 687-721
It should be there:

    @property
    def rx0_lo(self):
        """rx0_lo: Carrier frequency of RX1 path"""
        return self._get_iio_attr("altvoltage0", "RX1_LO_frequency", True)

    @rx0_lo.setter
    def rx0_lo(self, value):
        self._set_iio_attr("altvoltage0", "RX1_LO_frequency", True, value)

    @property
    def rx1_lo(self):
        """rx1_lo: Carrier frequency of RX2 path"""
        return self._get_iio_attr("altvoltage1", "RX2_LO_frequency", True)

    @rx1_lo.setter
    def rx1_lo(self, value):
        self._set_iio_attr("altvoltage1", "RX2_LO_frequency", True, value)

    @property
    def tx0_lo(self):
        """tx1_lo: Carrier frequency of TX1 path"""
        return self._get_iio_attr("altvoltage2", "TX1_LO_frequency", True)

    @tx0_lo.setter
    def tx0_lo(self, value):
        self._set_iio_attr("altvoltage2", "TX1_LO_frequency", True, value)

    @property
    def tx1_lo(self):
        """tx1_lo: Carrier frequency of TX2 path"""
        return self._get_iio_attr("altvoltage3", "TX2_LO_frequency", True)

    @tx1_lo.setter
    def tx1_lo(self, value):
        self._set_iio_attr("altvoltage3", "TX2_LO_frequency", True, value)

tfcollins, 2024-04-11:
You likely have an older driver version. This API was updated months back https://github.com/analogdevicesinc/pyadi-iio/commit/28917785e9a1e824420353e4062c76c0084bf9d2

v0.0.15 will contain the older driver API.

michaelkimmer, 2024-04-12:
Do I understand right that you broke the code with this commit?

> You likely have an older driver version. This API was updated months back [2891778](https://github.com/analogdevicesinc/pyadi-iio/commit/28917785e9a1e824420353e4062c76c0084bf9d2)
> 
> v0.0.15 will contain the older driver API.

tfcollins, 2024-04-13:
No. The actual driver API changed that pyadi calls into. v0.0.16+ assumes a driver from Kuiper 2022-R2 or newer.

closed 2024-04-15 by tfcollins