analogdevicesinc/pyadi-iio #581: Connection through Serial gives error but with Ethernet it works fine

VighneshBhatVibhatsu, 2024-07-01:
I'm trying to connect with AD9082 board when using ethernet connection everything works fine. But when trying to connect through USB tho it works fine on the IIO-oscilloscope tool. using the library throughs error telling that self._rxadc and self._ctrl is both NULL hence no channels found. The fpga im connecting to is the VCU118 board.

VighneshBhatVibhatsu, 2024-07-01:
When trying to connect to the serial port (when no device is connected through it) on IIO-OSCILLOSCOPE it works perfectly but when trying --> " adi.ad9081(uri="serial:"+port+",115200,8n1n") "  it gives error telling me that NONE type dosent have the attribute Channel. when checking it further found that the variable in the logic which is self._rxadc present in the AD9081.py file is showing as NONE. why is this occuring

tfcollins, 2024-07-01:
Our VC118 based reference designs do not support libiio over serial out of the box. Have you modified the userspace? If not, can you provide screenshots of IIO-Scope's connection Window actually connected over serial?

tfcollins, 2024-07-01:
libiio works the same way in both software packages. Please provide a screenshot of IIO-Scope's connection window with use of the serial context connected as requested. It will provide more details on the connection itself.

VighneshBhatVibhatsu, 2024-07-02:
sure i have attached 2 images of it connecting and graphing of it through the software

[image]

[image]

VighneshBhatVibhatsu, 2024-07-02:
and attached below is a seperate connection using the library with 
[image]

tfcollins, 2024-07-02:
Ok I see the problem now. You are using No-OS.

No-OS provides zero control over AD9081/AD9082 itself. It simply provides you access to the DMAs in the FPGA. You can see based on the devices listed. Pyadi-iio exposes controls of the IIO driver for AD9081/AD9082. Since the drivers, or even limited variants sometimes available in No-OS, are not available when pyadi-iio goes to enumerate them it fails. If you want to talk to the device with IIO you need to use the libiio API directly or create a simple class based off the rx_tx classes within pyadi-iio.

Using just the DMAs alone like in No-OS is not supported in the current pyadi-iio codebase.

If you want to control AD9081/AD9082 remotely, you will need to switch to the Linux driver.

-Travis

VighneshBhatVibhatsu, 2024-07-03:
>If you want to talk to the device with IIO you need to use the libiio API directly or create a simple class based off the rx_tx classes within pyadi-iio.

Ok thank you for this information could you give the link or explain  on how to use the libiio directly if possible.

With regards,
Vighnesh

tfcollins, 2024-07-03:
Here is the doc for the python bindings https://analogdevicesinc.github.io/libiio/v0.25/python/index.html

Please post on ez.analog.com if you need further support as this is not a pyadi-ion issue

VighneshBhatVibhatsu, 2024-07-04:
thank you for your support ill do as suggested by you.

with regards,
vighnesh

closed 2024-07-04 by VighneshBhatVibhatsu