analogdevicesinc/pyadi-iio #602: ad7606. Channel 0 for Timestamp

Miavu87, 2024-10-01:
Hello,
I am new to a development with AD7606 and I am trying to run the example given in the repository: "pyadi-iio/examples
/ad7606.py".

The example shows an error on the line "sc = ad7606.scale_available" because it tries to read the data from channel 0.

In my device, it assigns channel 0 to the Timestamp:

```
        IIO context has 2 devices:
                iio:device0: ad7606-8
                        9 channels found: 
                                timestamp:  (input)
                                voltage0:  (input)
                                3 channel-specific attributes found: 
  ...
```
On the other hand, if I run iio_info the channel Timestamp is number 8.

```
      IIO context has 2 devices:
              iio:device0: ad7606-8 (buffer capable)
                      9 channels found:
                              voltage0:  (input, index: 0, format: le:S16/16>>0)
                              3 channel-specific attributes found:
                                      attr  0: raw value: 32767
      ....
```

I fixed everything by making a new file "ad7606.py" and changing the line:

```
          @property
          def scale_available(self):
              """Provides all available scale settings for the AD7606 channels"""
              return self._get_iio_attr(self.channel[0].name, "scale_available", False)
```

Along the line: 

```
        @property
        def scale_available(self):
            """Provides all available scale settings for the AD7606 channels"""
            return self._get_iio_attr(self.channel[0].name, "scale_available", False)
```

Is this the only solution?

THANK YOU!!

Miavu87, 2024-10-01:
Sorry,
the solution is:
`return` self._get_iio_attr(self.channel[1].name, "scale_available", False)`

tfcollins, 2024-10-01:
Can you generate a context file for the device and provide it here?
They can be created through use of the [pytest-libiio module](https://tfcollins.github.io/pytest-libiio/master/cli_tools/) or with [iio-emu](https://github.com/analogdevicesinc/iio-emu/tree/main/tools).

Miavu87, 2024-10-03:
Hi,
I have continued working with my device and having problems with Device Tree and Udev. At this moment, the device isn't created and therefore iio_info doesn't work for me.

As soon as I solve it I will send you what you requested.

Thank you.

Miavu87, 2024-10-03:
Hi,
I was able to generate the XML file (attached): context_file.xml ==> The order of the channels is as expected.

I attach a file with the context from a Python script (with import iio): context_from_python.txt ==> Timestamp is channel 0

Thanks,
[Files_context.zip](https://github.com/user-attachments/files/17248485/Files_context.zip)

tfcollins, 2024-10-04:
I don't think you used the correct tool to generate the XML file. Note that you cannot use [iio-genxm](https://github.com/analogdevicesinc/libiio/blob/v0.23/tests/iio_genxml.c) from libiio itself. This is different than the tools provided by iio-emu or pytest-libiio

Miavu87, 2024-10-05:
Hi,
 i attached the file generated with command "gen_xml --uri local: --xml
generated.xml"

Seems to be OK, but in my IDE (VsCode) I can see channel 0 as Timestamp.

Thank you.

tfcollins, 2024-10-15:
In your files you can see the log messages:

```
XML generated:

<?xml ...

Context re-creation from generated XML succeeded!
```

Those surrounding log messages (not xml data) only come from iio-genxml. See here https://github.com/analogdevicesinc/libiio/blob/main/utils/iio_genxml.c#L99

Miavu87, 2024-10-16:
Hi,
I'm sorry but I don't know how to get the XML file from pytest-libiio or iio-emu.

Could you send me an example?

Thanks again.

tfcollins, 2024-10-16:
> Can you generate a context file for the device and provide it here? They can be created through use of the [pytest-libiio module](https://tfcollins.github.io/pytest-libiio/master/cli_tools/) or with [iio-emu](https://github.com/analogdevicesinc/iio-emu/tree/main/tools).

I detailed this here with these tools

Miavu87, 2024-10-22:
Hello,
I'm sorry but I am absolutely unable to compile ios-emu. I have tried thousands of things and thousands of tests.

If you like, we'll drop the matter.

Thanks for everything.

closed 2024-12-20 by tfcollins