analogdevicesinc/pyadi-iio #607: An issue with ADALM-Pluto QPSK Tx and Rx in Python

aligharamohammadi, 2024-10-16:
I am replicating my MATLAB Tx and Rx code for ADALM-Pluto in Python. I have successfully configured the transmitter in Python and I'm using my MATLAB code for the receiver.

Currently, I am generating QPSK-modulated symbols and transmitting them. However, on the receiver side, I only seem to receive one symbol at a time. All the symbols are present, but only one symbol appears to be captured per transmission.

Does anyone know what could be causing this issue? Could it be related to buffer size, or is there some synchronization or data capturing setting I'm missing?

tfcollins, 2024-10-16:
We'll need more information. Can you provide a code example?

aligharamohammadi, 2024-10-17:
[TxPluto.zip](https://github.com/user-attachments/files/17413651/TxPluto.zip)

This is my Tx code. For Rx, I use the Simulink example. I just adjusted the sampling rate. 

openExample('plutoradio/QPSKReceiverWithADALMPLUTORadioSimulinkExample')

tfcollins, 2024-10-17:
Your dataset is not scaled correctly for the device. Floats are directly cast to ints making all your data into 3 possible DAC codes [-1,0,1]. Data needs to be scaled between to the DAC fullscale of 2^15

-Travis

aligharamohammadi, 2024-10-17:
[TxPluto-Revised.zip](https://github.com/user-attachments/files/17417044/TxPluto-Revised.zip)
Thnaks. 
I have revised the code. Still, it is not solved completely.

tfcollins, 2024-10-17:
What are you observing and what do you expect to observe?

-Travis

aligharamohammadi, 2024-10-17:
The following plot is from Noise
[image]
I have attached my scatter plot here.
[image]
 I expect to see a qpsk modulation as following
[image]

tfcollins, 2024-10-17:
I'm not really sure what to tell you as you are looking at the end of a receive algorithm. The code you have generates data fine. If the TX side is working is MATLAB, how do they differ at the receiver before applying correction?

tfcollins, 2024-12-09:
Closing due to inactivity