analogdevicesinc/pyadi-iio #619: Imported library not executing

TeddySw, 2024-12-09:
I'm attempting to import the library using python 3.13.0 but I'm getting the following error: 

Traceback (most recent call last):
  File "C:\...\Pluto\Radio_Pulse_Gen.py", line 1, in <module>
    import adi
  File "C:\...\Lib\site-packages\adi\__init__.py", line 5, in <module>
    from adi.ad2s1210 import ad2s1210
  File "C:\...\Lib\site-packages\adi\ad2s1210.py", line 6, in <module>
    from adi.context_manager import context_manager
  File "C:\...\Lib\site-packages\adi\context_manager.py", line 5, in <module>
    import iio
  File "C:\...\Lib\site-packages\iio.py", line 227, in <module>
    _lib = _cdll(find_library(_iiolib), use_errno=True, use_last_error=True)
  File "C:\...\Lib\ctypes\__init__.py", line 380, in __init__
    if '/' in name or '\\' in name:
       ^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable

Any idea what I may be doing wrong?

tfcollins, 2024-12-09:
Did you install the C library for libiio? https://analogdevicesinc.github.io/pyadi-iio/guides/quick.html

tfcollins, 2024-12-20:
No response for a few weeks. Closing

closed 2024-12-20 by tfcollins