analogdevicesinc/pyadi-iio #622: USB not connecting despite manual

tekntrash, 2024-12-12:
At https://analogdevicesinc.github.io/pyadi-iio/guides/connectivity.html it says "_Devices that are connected over USB or are on a system with IIO devices like a ZC706 or Zedboard, should be able to automatically connect without defining a URI like:
# Import the library
import adi
# Create a device interface
sdr = adi.Pluto()

We are trying that code with a SDP board EVAL-SDP-CB1Z and we get 
_Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/openvla-env/lib/python3.10/site-packages/adi/rx_tx.py", line 734, in __init__
    rx_def.__init__(self, *args, **kwargs)
  File "/root/openvla-env/lib/python3.10/site-packages/adi/rx_tx.py", line 653, in __init__
    shared_def.__init__(self, *args, **kwargs)
  File "/root/openvla-env/lib/python3.10/site-packages/adi/rx_tx.py", line 615, in __init__
    raise Exception("No context could be found for class")
Exception: No context could be found for class_

tfcollins, 2024-12-13:
I don't understand what you are trying to do with those exact lines and your SDP based system. The adi.Pluto class is designed to interface with the ADALM-PLUTO or similar based AD936x based device.

What hardware are you attaching to the SDP kit? Are you running device drivers on the SDP?

SDP is also likely serial based which is different than USB in the context of libiio, which won't support autodiscovery.

tekntrash, 2024-12-13:
We are trying to connect it to an nvidia jetson orin

tfcollins, 2024-12-13:
I assume with your Windows setup you are using adi.ad2s120 class. How does your Windows vs. Linux script diff when addressing the device? What errors do you get? Does the kernel register the SDP device when plugged into your Jetson board?

-Travis

tekntrash, 2024-12-14:
With windows we are just using the software included, not programming anything. And it runs on a laptop. 

This time we need to code a software

tfcollins, 2024-12-17:
I would recommend opening a thread [here](https://ez.analog.com/data_converters/precision_adcs/f/q-a/) to verify the eval system you are using support the IIO drivers which pyadi-iio supports. Sometimes the eval software support custom firmware that is only supported by the custom GUIs.

tekntrash, 2024-12-17:
In all honesty, not wasting our times: we created 2 threads there already and no one answered. Thanks anyways

tfcollins, 2024-12-17:
Can you link those threads?

closed 2024-12-18 by tekntrash