analogdevicesinc/pyadi-iio #628: dds.__update_dds throws an exception

Josvth, 2025-01-24:
Hi! When running https://github.com/analogdevicesinc/pyadi-iio/blob/main/examples/ad9361_example.py using the ADRV9361-Z7035 on the FMC carrier the following exception is raised:

```
python3 ad9361_example.py 
RX LO 2000000000
Traceback (most recent call last):
  File "ad9361_example.py", line 43, in <module>
    sdr.tx(iq)
  File "/home/jhof/.local/lib/python3.8/site-packages/adi/rx_tx.py", line 481, in tx
    self.disable_dds()
  File "/home/jhof/.local/lib/python3.8/site-packages/adi/dds.py", line 60, in disable_dds
    self.dds_enabled = np.zeros(self._num_tx_channels * 2, dtype=bool)
  File "/home/jhof/.local/lib/python3.8/site-packages/adi/dds.py", line 100, in dds_enabled
    self.__update_dds("raw", value)
  File "/home/jhof/.local/lib/python3.8/site-packages/adi/dds.py", line 24, in __update_dds
    chan = self._txdac.find_channel("altvoltage" + str(indx), True)
  File "/usr/lib/python3.8/site-packages/iio.py", line 1257, in find_channel
    chn = _d_find_channel(self._device, name_or_id.encode("ascii"), is_output)
  File "/usr/lib/python3.8/site-packages/iio.py", line 61, in _check_null
    raise OSError(err, _strerror(err))
OSError: [Errno 0] Success
```

Debugging shows that `len(self._txdac.channels)` in https://github.com/analogdevicesinc/pyadi-iio/blob/c7c4ca6c5a7f7c0e977b58b6e29f8823355d46d5/adi/dds.py#L20 returns 12, were-as for the ad9361 the only available `altvoltage` signals are 0-7 causing https://github.com/analogdevicesinc/pyadi-iio/blob/c7c4ca6c5a7f7c0e977b58b6e29f8823355d46d5/adi/dds.py#L21 to throw the above exception.

tfcollins, 2025-01-24:
Looking at the error from the trace the behavior of the libiio bindings does not appear correct. What version of pylibiio and libiio (the C library) are you using?

Josvth, 2025-01-25:
Hey @tfcollins, my perma-links were not to the tagged branch I fixed that in my previous comment. I'm using the networked back-end from a host to a target, here's the versions I'm running let me know if that gives the information you need.

on the host:
```
[jhof@ngav-z6-769 pysdr]$ iio_info -V
iio_info version: 1.0 (git tag:8bce8b32)
Libiio version: 1.0 (git tag: 8bce8b32) backends: local ip usb xml
[jhof@ngav-z6-769 pysdr]$ pip list | grep iio
pyadi-iio           0.0.18
pylibiio            1.0
```

and on the target:
```
analog@analog:~ $ iio_info -V
iio_info version: 0.26 (git tag:bdd5c000)
Libiio version: 0.26 (git tag: bdd5c00) backends: local xml ip usb serial
```

tfcollins, 2025-01-25:
Can you downgrade to libiio v0.XX? libiio v1 is unstable and this commit has undesired behavior

Josvth, 2025-01-27:
Hi @tfcollins, I can confirm that when downgrading to libiio v0.26 and re-installing pyadi-iio the issue is resolved. Any indication what is causing the problem for v1.0? I can patch pyadi-iio for the time being with a try-expect statement around the line that throws an error, but I would rather know what is causing the actual issue of 12 instead of 8 channels being reported.

tfcollins, 2025-01-28:
Returning more than 8 channels is expected. The issue is that in v1 the find_channel method will generate errors instead of returning None when no channel with a specific name is found. This is a bug in the libiio v1 bindings. You can see the return error is 0 which is not a valid error code

closed 2025-02-10 by tfcollins