analogdevicesinc/pyadi-iio #637: tx_two object has no setter

rookiebeotch, 2025-02-14:
i have ver 0.18.
when i run:
sdr = adi.adrv9002(uri="local:")
i get this error:

`Traceback (most recent call last):                                             
  File "<stdin>", line 1, in <module>                                          
  File "/usr/local/lib/python3.11/dist-packages/adi/adrv9002.py", line 75, in _
_init__                                                                        
    self._tx2 = tx_two(self._ctx, self._txdac2, self._tx2_channel_names)       
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^       
  File "/usr/local/lib/python3.11/dist-packages/adi/obs.py", line 59, in __init
__                                                                             
    self._complex_data = complex_data                                          
    ^^^^^^^^^^^^^^^^^^                                                         
AttributeError: property '_complex_data' of 'tx_two' object has no setter
`

my device (adrv9002) has both channels enabled and split.

tfcollins, 2025-02-15:
Try the current commit from the main branch. I believe this was fixed by this commit awhile back https://github.com/analogdevicesinc/pyadi-iio/commit/a40cafbba55680830d7a0a213c623216c0530daa

tfcollins, 2025-02-19:
Release was created this morning.

Can you expand on what you mean by examples not being updated? The referenced example works just fine.

tfcollins, 2025-02-19:
ah ok yeah the digital_gain_control_mode_chan0 I think are only automatic and SPI now but need to look at the test code. In regards to the interface gains, these are specific to the part configuration. There is a way to see the valid ones based on the current part state like so:

```python

dev = adi.adrv9002()
chan = dev._ctrl.find_channel('voltage0', 'interface_gain_available', False)
print(chan.value)
```

csmithmitll, 2025-02-24:
Thanks for suggestion.  The code above didn't quite work.  It gave the following error:

```
Traceback (most recent call last):
  File "C:\Program Files (x86)\JetBrains\PyCharm Community Edition 2024.1.3\plugins\python-ce\helpers\pydev\_pydevd_bundle\pydevd_exec2.py", line 3, in Exec
    exec(exp, global_vars, local_vars)
  File "<input>", line 1, in <module>
TypeError: find_channel() takes from 2 to 3 positional arguments but 4 were given
```

I ended up setting the control mode to "spi" and using this to find an acceptable gain:
```
test=sdr._ctrl.find_channel('voltage0')
test.attrs["interface_gain_available"].value
'0dB'
```
Then everything after that worked.