analogdevicesinc/pyadi-iio #646: AD9081 wrong DAC data when multi-device

liangyiqiu, 2025-03-10:
I am using two AD9081-FMCA-EBZ and VCU118 boards with iio Linux drivers. When I connect a second device in the pyadi code, the DAC data of the second device will be wrong. However, the ADC data of both devices are fine. This phenomenon can be reproduced stably. 

For example, 
`dev1 = adi.ad9081("ip:192.168.2.1")`
`dev2 = adi.ad9081("ip:192.168.2.2")`
···
`dev1.tx([tx_waveform[:,0],tx_waveform[:,1],tx_waveform[:,2],tx_waveform[:,3]])`
`dev2.tx([tx_waveform[:,0],tx_waveform[:,1],tx_waveform[:,2],tx_waveform[:,3]])`

The two devices send the same data. The DAC output of dev1 is normal, as follows:

[image]

However, the DAC output of dev2 is wrong, as follows:

[image]

When connecting one of the devices separately, the DAC output is normal. 

I guess the problem is that the two devices share some resources in pyadi.

tfcollins, 2025-03-11:
Please provide the full script and version or commit used of pyadi-iio

liangyiqiu, 2025-03-11:
> Please provide the full script and version or commit used of pyadi-iio

Thank you! 

The version of pyadi-iio is 0.0.19. 

After more tests, I found that when there are two devices, only the first ADC data of dev2 is fine, and the other ADC and DAC data are wrong.

Here is a script to verify this issue：
```
import adi
import matplotlib.pyplot as plt
import numpy as np
from scipy import signal

dev1= adi.ad9081("ip:192.168.2.1") # If this line is deleted, everything works fine.
dev = adi.ad9081("ip:192.168.2.2")

# Configure properties
print("--Setting up chip")

# Total number of channels
N_RX = len(dev.rx_channel_nco_frequencies)
N_TX = len(dev.tx_channel_nco_frequencies)

# Total number of CDDCs/CDUCs
NM_RX = len(dev.rx_main_nco_frequencies)
NM_TX = len(dev.tx_main_nco_frequencies)

dev._ctx.set_timeout(3000)
dev._rxadc.set_kernel_buffers_count(1)
dev._txdac.set_kernel_buffers_count(2)

# Set NCOs
dev.rx_channel_nco_frequencies = [0] * N_RX
dev.tx_channel_nco_frequencies = [0] * N_TX

dev.rx_main_nco_frequencies = [1000000000] * NM_RX
dev.tx_main_nco_frequencies = [1000000000] * NM_TX

dev.jesd204_fsm_ctrl = "1"  # This will resync the NCOs and all links

dev.rx_main_6dB_digital_gains = [1] * NM_RX
dev.tx_main_6dB_digital_gains = [1] * NM_TX

dev.rx_channel_6dB_digital_gains = [1] * N_RX
dev.tx_channel_6dB_digital_gains = [1] * N_TX

dev.rx_enabled_channels = [0,1]
dev.tx_enabled_channels = [0,1,2,3]

dev.rx_nyquist_zone = ["odd"]

dev.rx_buffer_size = 2**19
dev.tx_cyclic_buffer = False

dev.disable_dds()

# load waveform
from scipy import signal
import scipy.io as scio

dataFile = '../waveform/tx_waveform.mat'
data = scio.loadmat(dataFile)
tx_waveform=(data['txWaveform'])

dev.tx_ddr_offload = 0

if dev.jesd204_device_status_check:
    print(dev.jesd204_device_status)

dev.rx_sync_start = "arm"
dev.tx_sync_start = "arm"

if not ("arm" == dev.tx_sync_start == dev.rx_sync_start):
    raise Exception(
        "Unexpected SYNC status: TX "
        + dev.tx_sync_start
        + " RX: "
        + dev.rx_sync_start
    )

dev.tx_destroy_buffer()
dev.rx_destroy_buffer()
dev._rx_init_channels()

dev.tx([tx_waveform[:,0],tx_waveform[:,1],tx_waveform[:,2],tx_waveform[:,3]])

dev.tx_sync_start = "trigger_manual"

if not ("disarm" == dev.tx_sync_start == dev.rx_sync_start):
    raise Exception(
        "Unexpected SYNC status: TX "
        + dev.tx_sync_start
        + " RX: "
        + dev.rx_sync_start
    )

try:
    x = dev.rx()
except Exception as e:
    print("Run#----------------------------- FAILED:", e)

# save rx waveform
dataNew = '../waveform/rx_waveform_0.mat'
scio.savemat(dataNew, {'rx_waveform':x[0]})

dataNew = '../waveform/rx_waveform_1.mat'
scio.savemat(dataNew, {'rx_waveform':x[1]})

print("Waveform saved. size:",len(x))
```
I also used a spectrum analyzer to observe the DAC output, and it was indeed an incorrect spectrum.

tfcollins, 2025-03-13:
I'm out of the office but I can take a look with hardware on Monday