analogdevicesinc/pyadi-iio #663: adxl355 device not found

niccolodematteis, 2025-04-15:
Dear developers,
I've got an ADXL355-PMDZ and successfully connected to a Raspberry-Pi 5. I've made some tests using this library https://github.com/gpvidal/adxl355-python and I obtained readings. However, after successful installing of pyadi-iio, when I run `acc = adi.axdl355()`, I get the error `No device found`.
Please, can you give me some help?
Thanks in advance

niccolodematteis, 2025-04-18:
Thank you @tfcollins. 
I'm not sure if I have correctly understood how to configure my system. I've found this instructions https://wiki.analog.com/resources/tools-software/linux-software/pyadi-iio
Accordingly, I need to
1) install libiio
2) install pylibiio
3) install pyadi-iio (I can't find where anymore, but I read that installing pyadi-iio using conda would install pylibiio automatically)

I don't understand where should I put/use the linux drivers.

Plus, I see that there is a raspberry-dedicated linux distro https://wiki.analog.com/resources/tools-software/linux-software/kuiper-linux that can be mounted on a SD card. I suppose this distro already contains all the libraries I need to manage the accelerometer (including linux drivers). But it is just an alternative(?)

tfcollins, 2025-04-18:
All the libiio related software components are designed to work with IIO based kernel drivers (https://wiki.analog.com/software/linux/docs/iio/iio) . Therefore, your kernel must of the related driver which must be activated through devicetree on your system.

Yes, Kuiper should contain a kernel that contains this driver which can be enabled through the RPI's config.txt plus the overlay. If you need help with that aspect please post a question here https://ez.analog.com/linux-software-drivers/ . We really only answer pyadi-iio specific questions here.