analogdevicesinc/pyadi-iio #664: EVAL-CN0554-RVIZ ERROR: READ LINE: -32

melE2a, 2025-04-29:
I am trying to test the operation of the EVAL-CN0554-RVIZ board. When it tries to change the value of the DAC after taking a reading of the ADC it returns the following error:

ERROR: READ LINE: -32
Traceback (most recent call last):
  File "/home/analog/pyadi-iio/examples/cn0554_example.py", line 78, in <module>
    my_cn0554.dac.voltage0.volt = 0
  File "/usr/local/lib/python3.9/dist-packages/adi/ltc2688.py", line 96, in volt
    self.raw = int(((val / self.scale) - self.offset) * (ltc2688.vref / 4.096))
  File "/usr/local/lib/python3.9/dist-packages/adi/ltc2688.py", line 45, in scale
    return self._get_iio_attr(self.name, "scale", True, self._ctrl)
  File "/usr/local/lib/python3.9/dist-packages/adi/attribute.py", line 127, in _get_iio_attr
    self._get_iio_attr_str(channel_name, attr_name, output, _ctrl)
  File "/usr/local/lib/python3.9/dist-packages/adi/attribute.py", line 122, in _get_iio_attr_str
    return channel.attrs[attr_name].value
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 703, in <lambda>
    lambda self: self._read(),
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 732, in _read
    _c_read_attr(self._channel, self._name_ascii, buf, len(buf))
  File "/usr/local/lib/python3.9/dist-packages/iio.py", line 62, in _check_negative
    raise OSError(-result, _strerror(-result))
BrokenPipeError: [Errno 32] Broken pipe

It is not a cable connection problem because we are in a preliminary testing phase.

This error already appears when I run the example code cn0554_example.py

AndyOesmer, 2025-04-30:
Hi @melE2a,

Can you kindly post the output of iio_info? And also what version of ADI Kuiper Linux are you using?

Thanks,
Andy

AndyOesmer, 2025-05-21:
Hi @melE2a ,
Please do not hesitate to post a question in this forum: [https://ez.analog.com/reference-designs/f/q-a] for support on the CN0554. Thanks!